pikvm/pikvm #1279: Mouse and keyboard do not connect when windows restarts

Deefury13, 2024-03-19:
When you restart windows both the mouse and keyboard turn orange and cease to function unless the PiKvm is rebooted

**To Reproduce**

1. restart windows

**Expected behavior**
Should reconnect when windows restarts

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: windows 11

**PiKVM info:**
  - Raspberry Pi board version RPi 4 B 8GB
  - PiKVM platform V3 pice version

**Additional context**
Add any other context about the problem here.

mdevaev, 2024-03-19:
Hello.

> PiKVM platform V3 pice version

What device do you mean? Please take a photo or give a link.

Deefury13, 2024-03-19:
Geekworm KVM-A8 PCIe Version for... https://www.amazon.com/dp/B0BN1MFHMW?ref=ppx_pop_mob_ap_share
[image]
[image]

mdevaev, 2024-03-19:
This is not PiKVM device. We only support our own devices and DIY builds. If you have problems with third-party devices, contact their manufacturer.

closed 2024-03-19 by mdevaev