pikvm/pikvm #1285: v4 DIY instructions

hryamzik, 2024-03-21:
Is there a way to build v4 mini yourself? v3 works fine but does miss passive cooling.

mdevaev, 2024-03-21:
Hello. The V4 is sold as a ready-to-use device complete with CM4 and case. There are no DIY options for this.

closed 2024-03-21 by mdevaev

hryamzik, 2024-03-25:
Does this mean hardware is not open sourced?

mdevaev, 2024-03-25:
The software is open, the hardware is not. It's our way of monetization, since PiKVM is our full time work. Thanks to the closed hardware, the software remains open and high quality.

hryamzik, 2024-03-25:
I totally get your points, was mostly interested in DYIing to make a silent unit but otherwise older versions are fine. Thanks for keeping software code open!

mdevaev, 2024-03-25:
Well, V4 Mini is already silent unit with passive cooling. We have selected and tested the most suitable parts for this.