pikvm/pikvm #1287: Two black edges which reduces screen real estate when connecting pc has a higher resolution than the host's resolution

plia7, 2024-03-23:
**Describe the bug**
When connecting to PiKVM's website through a browser from a connecting pc that has a higher resolution than the host's resolution it causes two black edges to show which reduces screen real estate. The issue is happening in both landscape and portrait orientation.

**Prerequisites:** Set your host to have a 1920x1080 resolution. Set your connecting pc display resolution to 2560x1600. 

**To Reproduce**
Steps to reproduce the behavior, like:
1. Open Chrome browser and connect to PiKVM.
2. Observe the host screen either in full screen or not.

**Actual behavior**
Two black edges show which reduce the screen real estate meaning the connecting pc display real estate is not being fully used. The mouse can't move to the the black edges areas.

**Expected behavior**
No two black edges are showing, host screen fills in full and the mouse can move to the areas where the black edges areas are currently showing.

**Workaround**
Change the connecting pc resolution to match the host resolution.

**Screenshots**
Two black edges show in landscape resolution 2560x1600:
[image]

Compared with no black edges in landscape resolution 1920x1080:
[image]

Two black edges show in portrait resolution 1600x2560:
[image]

Compared with no black edges in portrait resolution 1080x1920:
[image]

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
Windows 11
 - Browser [e.g. chrome, safari]
Google Chrome Version 123.0.6312.58 (Official Build) (64-bit)
 - Version [e.g. 22]
 - VNC client (if used)

**PiKVM info:**
  - Raspberry Pi board version [e.g. RPi 4]
PiKVM v3 Pre-Assembled
  - PiKVM platform [e.g. v2-hdmi]
  - Video capture type [e.g. CSI bridge]
  - KVMD version: `pacman -Q | grep kvmd`
kvmd 3.321-1
kvmd-fan 0.30-1
kvmd-oled 0.26-1
kvmd-platform-v3-hdmi-rpi4 3.321-1
kvmd-webterm 0.48-1
  - uStreamer version: `pacman -Q | grep ustreamer`
ustreamer 6.4-1
  - Linux kernel: `uname -a`
Linux pikvm 6.6.21-4-rpi #1 SMP Wed Mar 13 20:36:42 UTC 2024 armv7l GNU/Linux

**Additional context**
It's important to be able take full advantage of the connecting pc display real estate when viewing/controlling the host without the need to lower/sacrifice the connecting pc display resolution which is not the ideal solution or workaround because there could be other windows in the same screen/other tabs in the browser besides PiKVM tab that you want to see in the highest resolution possible of the connecting pc.

mdevaev, 2024-03-23:
Did you click this?
[image]

plia7, 2024-03-23:
> Did you click this? [...]

Yes, but this only works for non full screen mode and it just adjusts the rectangular around the host screen. It still doesn't get rid of the black edges when I'm in full screen when in higher resolution.

mdevaev, 2024-03-23:
The OS scales the resolution and adds these bands in case of different aspect ratio. You can't get rid of them. Use the native resolution.

plia7, 2024-03-24:
> The OS scales the resolution and adds these black egges in case of different aspect ratio. You can't get rid of them. Use the native resolution.

My screen is (2560 x 1600) IPS Monitor, 16:10 Aspect Ratio.

What is the aspect ratio for PiKVM 1080p resolution?

mdevaev, 2024-03-24:
> What is the aspect ratio for PiKVM 1080p resolution?

16:9

>  Can I change it in display advanced settings inside the host and/or EDID?

Aspect ratio is a resolution based factor. 1920x1080 has 16:9, 1920x1200 (V4 supports it) has 16:10.

> When you say to use native resolution are you saying that the connecting pc resolution should match the host resolution to get rid of the black edges as the solution?

Not resolution, just aspect ration.

Maybe I misunderstood something, take the whole video of the screen with browser.

plia7, 2024-03-24:
> The OS scales the resolution and adds these black egges in case of different aspect ratio. You can't get rid of them. Use the native resolution.

Ok I think you are right here, after doing a little bit more testing, I understand now why in this one HD monitor it appears to be full is because it's scaled at 100%.
When using a higher resolution monitor, and trying to set a lower resolution (HD) with 100% scaling (as the HD monitor), then it does the same black edges, and that's the same behavior outside the PiKVM, so I believe that's expected behavior.
I'm closing the ticket.

Thanks.

closed 2024-03-24 by plia7