pikvm/pikvm #1291: Split mouse/keyboard to usb hid and pico bridge

No0ne, 2024-03-28:
Looking at https://docs.pikvm.org/pico_hid_bridge/#the-final-steps both mouse and keyboard gets handed over to the pico hid.
But can you also split it up, so that the mouse gets routed thru the pi usb and the keyboard thru the pico ps/2?

Full issue see: https://github.com/No0ne/ps2x2pico/issues/29

mdevaev, 2024-03-28:
The current plugin interface does not allow this, there can only be one HID.

No0ne, 2024-03-29:
Ok, alternativly is it possible to connect a Pico HID via SPI on the PiKVM V4 board, e.g. by routing cables thru the wifi holes?

mdevaev, 2024-03-29:
V4 has no available SPI :(

No0ne, 2024-04-04:
Ok, thanks for that info, I'll try out the UART-way for this.

No0ne, 2024-04-06:
Thanks for the info, we got it working using the serial connection aka. GPIO22 jumpered to GND: https://github.com/No0ne/ps2x2pico/issues/29#issuecomment-2040496088

closed 2024-04-06 by No0ne

mdevaev, 2024-04-06:
Good to know :ok_hand:

JeWe37, 2024-05-15:
Hi @No0ne, what does your config look like in the end? I'm trying to do exactly this, have the mouse, usb drive etc. run through the OTG port and the keyboard through the pico HID. If I'm misunderstanding and you worked around this another way, could this issue be reopened? I would be interested in such functionality.

No0ne, 2024-05-15:
Hi, the main problem was that the pikvm v4 doesn't have SPI pins for the pico HID. We solved it by connecting via the serial interface as seen here: https://github.com/No0ne/ps2x2pico/issues/29#issuecomment-2040496088

What keyboard do you want to emulate, USB or PS/2?

JeWe37, 2024-05-15:
The issue is that I am trying to hook up the PiKVM to a traditional KVM. In order to switch it, I have to have a keyboard connected to its keyboard port, such that I can use the shortcuts to switch between PCs. That would be fine, except its dedicated keyboard port only supports keyboards, and no other peripherals. So to have other peripherals, I need to hook it up another USB port on it.

TLDR; I genuinely need 2 separate USB ports. I have a USB OTG port and a Pico HID, I want to combine them to be able to connect to two different USB ports.

No0ne, 2024-05-15:
You need to go thru PS/2 as shown here: https://github.com/No0ne/ps2x2pico/issues/29#issuecomment-2017771665
As mdevaev mentioned https://github.com/pikvm/pikvm/issues/1291#issuecomment-2026195470 its not possible to split keyboard and mouse right now so both have to go thru the pico.

JeWe37, 2024-05-15:
> You need to go thru PS/2 as shown here: [No0ne/ps2x2pico#29 (comment)](https://github.com/No0ne/ps2x2pico/issues/29#issuecomment-2017771665) As mdevaev mentioned [#1291 (comment)](https://github.com/pikvm/pikvm/issues/1291#issuecomment-2026195470) its not possible to split keyboard and mouse right now so both have to go thru the pico.

I see, though that won't work in my case. But given that that is very much a workaround. Therefore I'd appreciate if this could be reopened. I will perhaps see how complicated it would be to allow for multiple HID interfaces depending on the device and submit a PR if I can figure it out.