pikvm/pikvm #1293: Add Support for Arbitrary Keyboard Layouts to Text Paste Function in Pi-KVM

rpdelaney, 2024-03-29:
The current implementation of the text paste function in PiKVM supports a limited set of keyboard layouts. This limitation can be an accessibility obstacle for users who rely on non-standard keyboard layouts. The feature request is to extend the text paste function to support arbitrary user-defined keyboard layouts.

Adding support for arbitrary keyboard layouts will significantly improve the accessibility of PiKVM, particularly for users who rely on non-standard keyboard layouts due to personal preference, accessibility needs, or regional requirements. This feature will enhance user experience and satisfaction, making PiKVM more inclusive and user-friendly.

## Proposed Solution

The proposed solution is to develop a mechanism within the text paste function that can allow users to configure their own keyboard layouts. 

## Alternatives Considered

An alternative solution could be to expand the list of supported keyboard layouts in the text paste function. However, this approach may not cover all possible keyboard layouts and would require continuous updates to accommodate new layouts. Therefore, a more flexible and scalable solution is to support user self-service.

GCustom, 2024-12-06:
This affects me as well, I run Dvorak on my workstation and the remote servers, but must switch the system controlled by my PiKVM v4 to QWERTY when using paste, and then switch it back before typing anything else besides a handful of keys like space or ctrl

mdevaev, 2025-11-16:
Okay, I see. I am gradually redesigning the keyboard subsystem and hope to solve some of these problems.