pikvm/pikvm #1294: Pinch To Zoom On Mobile

ThePieMonster, 2024-03-29:
**Describe the bug**
Pinch to zoom is sending the touch inputs to the remote device when in "KVM mode".

**To Reproduce**
Steps to reproduce the behavior, like:

1. Go to your PiKVM webpage.
2. Select KVM to remote into the machine.
3. Try to use two finger pinch to zoom commands.

**Expected behavior**
The screen will zoom in or out based on the touch input. 

**Desktop (please complete the following information):**
 - Browser - Android Chrome

**PiKVM info:**
  - Raspberry Pi board version [PiKVM 4 Plus] 
Used default settings for everything.

**Additional context**
Enabling "desktop mode" in the Android chrome browser will let you pinch to zoom in the non-browser space. 
You still cannot use pinch to zoom in the remote browser. 
If you zoom in to far then you can not zoom out as all that would be in view of the screen is the remote browser. 

Perhaps reading something like this may help for PiKVM frontend development: 
https://stackoverflow.com/questions/12392761/how-to-enable-pinch-zoom-on-website-for-mobile-devices

vieacq, 2024-05-04:
**+1**
...same config, except:
**PiKVM info:**
- PiKVM V3
**Remark:**
- There is no obvious alternate method to zoom in on the remote screen.
- Zoom in on remote screen is required due to the small size of displays on mobile devices

plia7, 2024-05-24:
Seems like a duplicate of the ticket that I created on Jul 28, 2023:
https://github.com/pikvm/pikvm/issues/1075

mdevaev, 2024-05-26:
Yes, it looks like a duplicate. Please subscribe to this issue to track it: https://github.com/pikvm/pikvm/issues/1075

closed 2024-05-26 by mdevaev