pikvm/pikvm #1296: I got a issue "Mouse and keyboard not have click response"

bi7jta, 2024-04-01:
**Describe the bug**
After boot use the Web Power menu , if I not have click activity long time since bootup, the mouse and keyboard will not accept my click request , till I disconnect and connect USB line between Raspberry Pi TypeC and PC USB.

My problem is : why will cause this ? Can I do some operation to activity piKVM mouse and keyboard again, use software?

Not one time , all bootup if I not wait the Windows 10 UI appear , will cause this .

Thank you very much! 

**Screenshots**
 
[image]

[image]

mdevaev, 2024-04-01:
Hello. Cursor is working?

mdevaev, 2024-04-03:
No feedback, closing.

closed 2024-04-03 by mdevaev