pikvm/pikvm #1300: Host reboot and getting into BIOS

mikrofyr, 2024-04-10:
**Describe the bug**
Problem to get keyboard and mouse to work during restart of host.
Have to click the "reset HID" button to get mouse/keyboard connection back after host restart.
If i time it right I can get it available to press del during POST for BIOS.
The mouse/keyboard icon turs red during host restart that indicates connection lost.

Zero issues with video.

**To Reproduce**
1. Establish connection to piKVM in browser
2. Restart host
3. Press del repeatedly to go into BIOS
4. Keyboard does not work, try again

**Expected behavior**
Keyboard shoud work at host reboot.
Workaround is "reset hid" with correct timing

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Windows and Linux
 - Browser [e.g. chrome, safari]

**PiKVM info:**
   - https://docs.pikvm.org/v1/
  - Linux kernel: Linux pikvm 6.6.21-4-rpi

**Additional context**
I have seen similar issues reported, but not sure if it is the same as this.
This bug makes the device inconveniant to use, but not a blocker.

PS: Really nice device overall

arch1mede, 2024-04-11:
Hi, can you join [Discord](https://discord.gg/bpmXfz5) to discuss further? Please mention this issue.

chchia, 2024-04-14:
most of the linux can simple do this command to restart to UEFI
`systemctl reboot --firmware-setup`

windows have different approach but it is able to boot to UEFI from windows as well.

arch1mede, 2024-05-06:
Hi, have you resolved this issue yet or joined [Discord](https://discord.gg/bpmXfz5) to discuss this further?