pikvm/pikvm #1305: Video is not streaming to remote network

rgov, 2024-04-20:
**Describe the bug**
I have a PiKVM attached to a Fedora system in a remote location. The PiKVM is connected to the LAN, and can reach the Internet. Tailscale is running on the PiKVM, too.

When I try to access to PiKVM stream over Tailscale, I get no video output, regardless of choosing WebRTC or HTTP streaming. In WebRTC mode, I just see a black image and usually "Inactive". In HTTP mode I see `<NO SIGNAL>`.

However, if I remote into a second system on the same network as the PiKVM using TeamViewer, and pull open the PiKVM web UI on that system, I *do* see the Fedora system video output.

**Desktop:**
 - OS: macOS
 - Browser: Safari
 - Version: 14.4
 - VNC client: None

**PiKVM info:**
  - Raspberry Pi board version: RPi CM 4
  - PiKVM platform: V4 Mini
  - Video capture type: (as above)
  - KVMD version:
    ```
    kvmd 3.329-1
    kvmd-fan 0.30-1
    kvmd-oled 0.27-1
    kvmd-platform-v4mini-hdmi-rpi4 3.329-1
    kvmd-webterm 0.48-1
    ```
  - uStreamer version: `ustreamer 6.9-1`
  - Linux kernel: `Linux pikvm 6.6.21-4-rpi #1 SMP Wed Mar 13 20:36:42 UTC 2024 armv7l GNU/Linux`

rgov, 2024-04-20:
I am able to access the video stream over TigerVNC.

arch1mede, 2024-04-23:
Hi, please join [Discord](https://discord.gg/bpmXfz5) to discuss further, please mention this issue when you join.

mdevaev, 2024-05-01:
This is clearly not a bug, but an installation problem, please use discord for support (if you haven't done it yet)

closed 2024-05-01 by mdevaev