pikvm/pikvm #1306: Improve guide for XH-HK4401

Ultranium, 2024-04-21:
I'd like to share my experience using what seems to be a slightly newer version of the XH-HK4401 4-port switch.

The main difference is it uses 3.3V non-inverted UART protocol for the remote control. 
This is actually great, because it eliminates need of using an inverting FTDI adapter, and also allows to connect the switch control interface directly to Pi's UART without using any extra circuitry. 

The markings on PCBs are:
- HDMI switch board: `PCB-HK4501-V0.2-A1-HDMI`
- USB/Control board: `PCB-HK4401-V2.0-B-USB`

Connecting this switch using the [current guide for the XH-HK4401](https://docs.pikvm.org/xh_hk4401/#switching-between-hosts-in-the-ui) won't work because of the different control interface implementation, so here are some instructions on how to do it.
I think, adding this information to the docs may help people with newer HK4401 switches.

**Before interfacing the switch with Raspberry using the following guide, one should first check the voltage levels on the control interface, because the manufacturer can change the implementation at any moment!!! If you accidentally connect a 5V version to your Pi, you can permanently damage it!**

### Using with FTDI adapter
I successfully used a $1 fake FTDI adapter from Aliexpress which looks like this:
<details>
<summary>FTDI adapter</summary>

[image]
</details>

1. Switch the jumper on the FTDI adapter to the position `3.3V`
2. Connect wires:
  - GND to USB GND (and shield, if present on cable)
  - TX to USB D- (white)
  - RX to USB D+ (green)
4. Remount the filesystem in read-write mode using the `rw` command
5. Append the following lines to `/etc/kvmd/override.yaml`:
<details><summary>override.yaml</summary>

```
kvmd:
    gpio:
        drivers:
            hk:
                type: xh_hk4401
                protocol: 2
                device: /dev/ttyAMA0
        scheme:
            ch0_led:
                driver: hk
                pin: 0
                mode: input
            ch1_led:
                driver: hk
                pin: 1
                mode: input
            ch2_led:
                driver: hk
                pin: 2
                mode: input
            ch3_led:
                driver: hk
                pin: 3
                mode: input
            ch0_button:
                driver: hk
                pin: 0
                mode: output
                switch: false
            ch1_button:
                driver: hk
                pin: 1
                mode: output
                switch: false
            ch2_button:
                driver: hk
                pin: 2
                mode: output
                switch: false
            ch3_button:
                driver: hk
                pin: 3
                mode: output
                switch: false
        view:
            table:
                - ["#Input 1", ch0_led, ch0_button]
                - ["#Input 2", ch1_led, ch1_button]
                - ["#Input 3", ch2_led, ch2_button]
                - ["#Input 4", ch3_led, ch3_button]

```
</details>

6. Remount the filesystem back to read-only mode using the command `ro` 
7. Run `systemctl restart kvmd`

### Using with direct UART connection
1. Remount the filesystem in read-write mode using the `rw` command 
2. Remove the `console=serial0,115200` parameter from `/boot/cmdline.txt`
3. Append the following lines to `/etc/kvmd/override.yaml`:
<details><summary>override.yaml</summary>

```
kvmd:
    gpio:
        drivers:
            hk:
                type: xh_hk4401
                protocol: 2
                device: /dev/ttyAMA0
        scheme:
            ch0_led:
                driver: hk
                pin: 0
                mode: input
            ch1_led:
                driver: hk
                pin: 1
                mode: input
            ch2_led:
                driver: hk
                pin: 2
                mode: input
            ch3_led:
                driver: hk
                pin: 3
                mode: input
            ch0_button:
                driver: hk
                pin: 0
                mode: output
                switch: false
            ch1_button:
                driver: hk
                pin: 1
                mode: output
                switch: false
            ch2_button:
                driver: hk
                pin: 2
                mode: output
                switch: false
            ch3_button:
                driver: hk
                pin: 3
                mode: output
                switch: false
        view:
            table:
                - ["#Input 1", ch0_led, ch0_button]
                - ["#Input 2", ch1_led, ch1_button]
                - ["#Input 3", ch2_led, ch2_button]
                - ["#Input 4", ch3_led, ch3_button]

```
</details>

5. Reboot
6. Connect wires:
  - GPIO pin 6 to USB GND (and shield, if present on cable)
  - GPIO pin 8 to USB D- (white)
  - GPIO pin 10 to USB D+ (green)

arch1mede, 2024-05-11:
Hi, Thank you for this feedback, This has been added to the existing documentation as additional follow-up. This issue will be closed.

closed 2024-05-14 by mdevaev

empty2k, 2024-05-26:
Hi Ultranium I bought the same HDMI switch, but I have a different FTDI adapter. I bought this one here. 
[image]
[image]

Unfortunately, I haven't been able to control the switch via the GPIO using the PiKVM so far. Would it be possible for you to show your setup? How does the wiring look from the switch to the PiKVM? A couple of photos would be appreciated.

Best regards and thank you

Ultranium, 2024-05-26:
Hi @empty2k .

You're using the UART adapter the wrong way.

You should plug it into a Pi's USB port, and then connect wires from adapter pins to the HDMI switch micro-USB control port like this (adapter side <-> switch side):

GND to USB GND (and shield, if present on cable)
TXD to USB D- (white)
RXD to USB D+ (green)

And no wires need to be connected to Pi GPIO.

My current setup is different from the reference one (because I use a custom solution to connect multiple ATX control boards), so it wouldn't be relevant here.

empty2k, 2024-05-26:
Wonderful! I must have mixed up a few things. I didn't really understand that I had to cut a Micro or Mini USB cable to connect the wires accordingly. However, I had to use /dev/ttyUSB0 in /etc/kvmd/override.yaml instead of /dev/ttyAMA0.
Apparently, I was lucky and nothing is damaged; at least, I haven't noticed any limitations so far.

Thank you very much @Ultranium

Ultranium, 2024-05-26:
Awesome!

You are absolutely right about using the `/dev/ttyUSB0`. It seems I forgot to change it in FTDI config in my manual. It's fixed now.

Ultranium, 2024-06-04:
> hi , i have: PCB-HK4501-V0.1-A-HDMI PCB-HK4401-V1.2-B-USB version and the method you mentioned not worked on me with cheap ftdi adapter same as yours. do you have any suggestion?
Hi @serkansmg.

Well, the best thing to do is to monitor exchange between the KVM switch and the remote first (using an oscilloscope or at least a multimeter), to find out what type of UART implementation is used on your particular unit.

If it's inverted (meaning there is about ~0V in idle mode on RX and TX lines), then you have to use an inverter IC the way it is described [here](https://docs.pikvm.org/xh_hk4401/#an-inverter-circuit-the-cheap-way).

If it is not inverted (there is ~3.3V or ~5V on RX and TX), set the FTDI adapter voltage switch accordingly, check your wiring and try to play with protocol version in `override.yaml` (line `protocol: 2`, possible values are 1 and 2).

serkansmg, 2024-06-04:
hi thanks for the fast reply.
on multimeter, placing probes to gnd-tx, gnd-rx, rx-tx it is not 0 on idle time. always there is ~3.2v
so is that mean it is not inverted?

also i replaced rx tx 
because on your document it is:
GND to USB GND (and shield, if present on cable)
TXD to USB D- (white)
RXD to USB D+ (green)

and on original document https://docs.pikvm.org/xh_hk4401/ it is like:

D- | White | RX
D+ | Green | TX
Gnd | Black | GND

i also switch +5v and 3.3v nothing changed.
also echo "SW1\r\nG01gA" > /dev/ttyUSB0 did not made any change (1-2-3-4)

Ultranium, 2024-06-04:
Well, it seems to me that your FTDI adapter may be broken.
Since you found out that your KVM switch uses 3.3V non-inverted UART, you can try to connect it to Pi's GPIO directly (see the "Using with direct UART connection" section in my original post).

If it doesn't work even via direct connection, I recommend you to get an oscilloscope with UART decoder or a logic analyzer (you can get these for just a few bucks on Aliexpress nowadays) for further debugging, just to be sure that the signal is being sent from Raspberry indeed. You can also then compare data exchange between Pi and KVM switch to the data stream between KVM switch and the original remote to find discrepancies.

serkansmg, 2024-06-04:
today i will try with waveshare usb-ttl adapter. i guess it is not a fake one.. 
anyway if it not works. 
i will try another interesting way.. 
i will use an esp32 and wire buttons on the controller's push buttons to esp32 gpio pins. :)) then from serial port on pikvm i will make it work to send commands to it.. 
so how will i configure override.yaml to send console commands?

Ultranium, 2024-06-04:
Wow, that's quite a complication!

I'm no expert on PiKVM extension, but I think you may need to write your own kvmd driver for that.
You could use the XH-HK4401 driver as a reference: https://github.com/pikvm/kvmd/blob/master/kvmd/plugins/ugpio/xh_hk4401.py

But still, I think it would be easier to troubleshoot the conventional solution for HK4401 using FTDI adapter or GPIO pins.

serkansmg, 2024-06-04:
esp32 is my area :)) i can more control over it.. 
anway, i'll check the driver. 
for the gpio , it did not worked for the ftdi adapter on pi..

serkansmg, 2024-06-04:
ok now i guided your suggestion and used directly raspberry pi pins.

` import RPi.GPIO as GPIO
 2 import time
 3
 4 # Define the GPIO pins connected to the push button inputs on the third-party board
 5 button_pins = [17, 27, 22, 23]
 6
 7 # Function to simulate a push button press
 8 def press_button(pin):
 9     GPIO.output(pin, GPIO.LOW)  # Simulate button press (active LOW)
10     time.sleep(0.05)            # Hold for 50 milliseconds (debounce time)
11     GPIO.output(pin, GPIO.HIGH) # Release button
12     time.sleep(0.05)            # Additional delay to prevent multiple triggers
13
14 # Set up the GPIO pins
15 GPIO.setmode(GPIO.BCM)
16 for pin in button_pins:
17     GPIO.setup(pin, GPIO.OUT, initial=GPIO.HIGH)
18
19 try:
20     while True:
21         for pin in button_pins:
22             press_button(pin)
23             time.sleep(2)  # Wait for 1 second before pressing the next button
24 finally:
25     GPIO.cleanup()`

this works and shows controller and device pins changing. 
how to reflect this to override.yaml?

what i tried based on gpio document:

`kvmd:
  gpio:
    drivers:
      __gpio__:  # Using the reserved name for the default GPIO driver
        type: gpio

    scheme:
      button1:
        driver: __gpio__
        pin: 17
        mode: output
        initial: true
        pulse:
          delay: 0.05
      button2:
        driver: __gpio__
        pin: 27
        mode: output
        initial: true
        pulse:
          delay: 0.05
      button3:
        driver: __gpio__
        pin: 22
        mode: output
        initial: true
        pulse:
          delay: 0.05
      button4:
        driver: __gpio__
        pin: 23
        mode: output
        initial: true
        pulse:
          delay: 0.05

    view:
      header:
        title: GPIO Controls
      table:
        - ["#Button Controls"]  # Label
        - ["Button 1", "button1"]
        - ["Button 2", "button2"]
        - ["Button 3", "button3"]
        - ["Button 4", "button4"]
`

makes pikvm gives nginx error and not let login

Ultranium, 2024-06-04:
Your override.yaml seems to be incorrect, it shouldn't have multiple GPIO sections.

Take a look at this file for a reference: https://github.com/shiecldk/pikvm-gpio-atx-multiport/blob/main/etc/kvmd/override.yaml

By the way, haven't you try to wire the KVM switch control port directly to the GPIO pins, like I told you? It could be a way easier solution if it works - it's only 3 wires for both sending command and receiving active channel #.

serkansmg, 2024-06-05:
i tried the direct gpio pins but not worked. but that gave me an idea :))
i directly soldered the controller unit's buttons with jumper cables and connect them to pi's gpios, each right pin was gnd, so i connected only one of it to pi's gnd (yellow cable in the picture). 
[image]
[image]
[image]

then i enabled command feature on pikvm (it is important, or the commands will not work)

```bash
nano /etc/sudoers.d/custom_commands
#added this
kvmd ALL=(ALL) NOPASSWD: ALL
```

then 
i wrote a python script:
# nano /root/gpio_control.py

```python
import RPi.GPIO as GPIO
import time
import sys

def usage():
    print("usage: gpio_control.py <pin>")
    sys.exit(1)

def press_button(pin):
    GPIO.output(pin, GPIO.LOW)  # Simulate button press (active LOW)
    time.sleep(0.05)            # Hold for 50 milliseconds (debounce time)
    GPIO.output(pin, GPIO.HIGH) # Release button
    time.sleep(0.05)            # Additional delay to prevent multiple triggers

def main():
    if len(sys.argv) != 2:
        usage()

    pin = int(sys.argv[1])

    GPIO.setmode(GPIO.BCM)
    GPIO.setup(pin, GPIO.OUT, initial=GPIO.HIGH)

    try:
        press_button(pin)
        time.sleep(2)
    finally:
        GPIO.cleanup()

if __name__ == "__main__":
    main()

```

then i created a basic bash script

# GPIO Control Shell Script

```bash
#!/bin/bash
# Shell script to control GPIO pins using a Python script

if [ "$#" -ne 1 ]; then
    echo "Usage: $0 <pin>"
    exit 1
fi

PIN=$1

/usr/bin/sudo /usr/bin/python3 /root/gpio_control.py $PIN
```

# Override Configuration for kvmd

```yaml
# nano /etc/kvmd/override.yaml
kvmd:
  gpio:
    drivers:
      input1:
        type: cmd
        cmd: [/usr/bin/sudo, /root/gpio_control.sh, 27]
      input2:
        type: cmd
        cmd: [/usr/bin/sudo, /root/gpio_control.sh, 17]
      input3:
        type: cmd
        cmd: [/usr/bin/sudo, /root/gpio_control.sh, 23]
      input4:
        type: cmd
        cmd: [/usr/bin/sudo, /root/gpio_control.sh, 22]

    scheme:
      input1-button:
        driver: input1
        pin: 0
        mode: output
        switch: false
      input2-button:
        driver: input2
        pin: 0
        mode: output
        switch: false
      input3-button:
        driver: input3
        pin: 0
        mode: output
        switch: false
      input4-button:
        driver: input4
        pin: 0
        mode: output
        switch: false

    view:
      header:
        title: GPIO Controls
      table:
        - []
        - ["#Click input button below to switch"]
        - []
        - ["input1-button|1 - Button 1", "input2-button|2 - Button 2", "input3-button|3 - Button 3", "input4-button|4 - Button 4"]

```

then restarted kvmd deamon

```bash
systemctl restart kvmd
```
voila... 
[image]

works perfectly.. 

all done...

Ultranium, 2024-06-05:
Well, what can I say... Wow!

But still, I found it strange that the original solution didn't work in your case. 
I kind of feel sorry that you had to go through all this, because my experience with PiKVM and XH-HK4401 went way smoother (although, I also had to spend a couple of days just to find out that the initial schematics with ICs or inverting FTDI adapters is redundant for newer versions of the HDMI switch).

aqxa1, 2024-07-16:
Does the USB to TTL adapter need to be FTDI (or clone) on the newer XH-HK4401? Given that the inverting isn't needed.

Since I already have a ch341 converter with 3.3V support.

thecode, 2024-07-25:
> Does the USB to TTL adapter need to be FTDI (or clone) on the newer XH-HK4401? Given that the inverting isn't needed.
> 
> Since I already have a ch341 converter with 3.3V support.

CH341 with 3.3V should work fine, I suggest to measure voltage & verify that your both your `XH-HK4401` and the converter are 3.3V, if they are it will work.

bluewhitep, 2025-01-11:
My setup is working with the following configuration:
- FTDI Device: DSD TECH SH-U09C
- Wiring:
  - TX → USB D− (white)
  - RX → USB D+ (green)
- Reasoning: Because the KVM’s TX and RX idle around 3.2V, the following settings are used:
  - No inversion on TX and RX
  - FTDI set to 3.3V

In operation, the RX LED blinks every 1–2 seconds.
When running `echo "SW3\r\nG03gA" > /dev/ttyUSB0` the TX LED flashes once.

FuNK3Y, 2025-02-15:
Is the connection communication bi-directional? Does pressing a physical button on the XH-HK4401 generate an event that can be caught by the pi kvm?

thecode, 2025-02-15:
> Is the connection communication bi-directional? Does pressing a physical button on the XH-HK4401 generate an event that can be caught by the pi kvm?

Yes, it works bidirectional, the web UI track the selected input even if changed from the physical button:
https://github.com/pikvm/kvmd/blob/02740aef374a2914cf85ba1cad16f5e93d0650f1/kvmd/plugins/ugpio/xh_hk4401.py#L178

maxpaine69, 2025-08-15:
so sorry to awaken this post, im trying to do something like described in this post.

- i have a swithc like that
- i already have a usb cable connected to it on the control port, and im receiving on the white and green cable 3.3v.

My question are is it possible to connect directly to the Pi usint the uart gpio pins?
I also read that its possible to get feedback back from the selection im i right?

serkansmg, 2025-08-16:
> so sorry to awaken this post, im trying to do something like described in this post.
>
> * i have a swithc like that
> * i already have a usb cable connected to it on the control port, and im receiving [...]

sure you can directly connect it to pi.

giautm, 2026-10-15:
If you have a control board like me here (v1.4), the protocol is just RS-232, don't need custom inverted RS-232. Any RS-232 with micro-USB cable can be used to control the switch.

[image]