pikvm/pikvm #1309: Update failing

kaystrobach, 2024-04-25:
**Describe the bug**

I have the kickstarter version of piKVM and realized, that it's time for an update.
Here is my journey and the request for help:

**To Reproduce**

use the kickstarter firmware and try to update it from that point:

```
    7  pikvm-update
    8  curl https://files.pikvm.org/update-os.sh | bash
    9  pikvm-update
   10  pacman-key --init
   11  pikvm-update
   12  datetime
   13  dae
   14  date
   15  gpg --refresh-keys
   16  sudo pacman-key --refresh-keys
   17  pikvm-update
```

**Expected behavior**

```
 curl https://files.pikvm.org/update-os.sh | bash
```

should lead to an uptodate system

**Screenshots**

I always get this and similar errors:

```
error: wireless-regdb: signature from "Arch Linux ARM Build System <builder@archlinuxarm.org>" is unknown trust
:: File /var/cache/pacman/pkg/wireless-regdb-2024.01.23-1-any.pkg.tar.xz is corrupted (invalid or corrupted package (PGP signature)).
Do you want to delete it? [Y/n]
```

**PiKVM info:**
  - Kickstarter version
  
**Additional context**
Add any other context about the problem here.

```
pacman-key --populate
sudo pacman-key --populate archlinux
```

all that results in

```

mdevaev, 2024-04-25:
Duplicate of #1310

closed 2024-04-25 by mdevaev

kaystrobach, 2024-04-25:
Yes done after quite some research.