pikvm/pikvm #1310: Update impossible

kaystrobach, 2024-04-25:
**Describe the bug**

I have the kickstarter version of piKVM and realized, that it's time for an update.
Here is my journey and the request for help:

**To Reproduce**

use the kickstarter firmware and try to update it from that point:

```
    7  pikvm-update
    8  curl https://files.pikvm.org/update-os.sh | bash
    9  pikvm-update
   10  pacman-key --init
   11  pikvm-update
   12  datetime
   13  dae
   14  date
   15  gpg --refresh-keys
   16  sudo pacman-key --refresh-keys
   17  pikvm-update
```

**Expected behavior**

```
 curl https://files.pikvm.org/update-os.sh | bash
```

should lead to an uptodate system

**Screenshots**

I always get this and similar errors:

```
error: wireless-regdb: signature from "Arch Linux ARM Build System <builder@archlinuxarm.org>" is unknown trust
:: File /var/cache/pacman/pkg/wireless-regdb-2024.01.23-1-any.pkg.tar.xz is corrupted (invalid or corrupted package (PGP signature)).
Do you want to delete it? [Y/n]
```

**PiKVM info:**
  - Kickstarter version
  
**Additional context**
Add any other context about the problem here.

```
pacman-key --populate
sudo pacman-key --populate archlinux
```

all that results in

```
==> ERROR: Trust database could not be updated.
```

I'm out of ideas:

```
==============================================================
      An unexpected error occurred during the update.
==============================================================

==============================================================
Please note that the filesystem now remain in Read-Write mode.
       A reboot is necessary to make it Read-Only again.
The reboot can be performed later using the 'reboot' command.
==============================================================
```

kaystrobach, 2024-04-25:
ok found this:

* `pacman-key --keyserver hkps://hkps.pool.sks-keyservers.net -r 912C773ABBD1B584` in https://github.com/pikvm/pi-builder/issues/16

mdevaev, 2024-04-25:
So is it solved now?

kaystrobach, 2024-04-25:
Yes should be mentioned in the docs regarding updates maybe.

closed 2024-04-25 by kaystrobach

TheKigen, 2024-05-18:
I just ran into this issue.

The better command to run in this case appears to be:
`sudo pacman -Syy gnupg archlinux-keyring`

Then you can rerun the update-os bash script.

@mdevaev I think it would be a good idea to update the keyring first in that update-os bash script before trying to update anything else from pacman.

mdevaev, 2024-05-19:
Maybe, but I'm not sure, updating gnupg might break something if gnupg is a new version and pacman is an old one.