pikvm/pikvm #1312: GPIO menu not auto closed

markfrancisonly, 2024-04-27:
**Describe the bug**

GPIO menu stays open after clicking menu item. Running a frequently used GPIO command now requires three clicks: 1) Open the GPIO menu, 2) Click to select an item, and 3) Click away to close menu ( 50% greater click burden (and I have bad hands ... )

**Expected behavior:**

- Auto close GPIO menu after item selection, same as other menus, unless otherwise specified. 

**Related enhancement idea for further improvement**

- Enable top-level GPIO items that are displayed fulltime and only require require a single click. Would be great to have PC channel selection top-level menu items, or other cmds. I frequently use a CTRL-ALT-DEL cmd, much more often than say Macros, currently three clicks away ... 
```
echo -ne "\x5\0\x4c\0\0\0\0\0" > /dev/kvmd-hid-keyboard
echo -ne "\0\0\0\0\0\0\0\0" > /dev/kvmd-hid-keyboard

```
 
**Desktop (please complete the following information):**
 - Browser: chrome 124

**PiKVM info:**
  -  kvmd 3.333-1 

**Config**
```
kvmd: 
    gpio:
        drivers:
            reboot:
                type: cmd
                cmd: [/usr/bin/sudo, reboot]
            restart_kvmd:
                type: cmd
                cmd: [/usr/bin/sudo, systemctl, restart, kvmd] 
            switch_input_ch1:
                type: cmd
                cmd: [/usr/bin/bash, /etc/kvmd/scripts/switch_input.sh, 1]
            switch_input_ch2:
                type: cmd
                cmd: [/usr/bin/bash, /etc/kvmd/scripts/switch_input.sh, 2]
            switch_input_ch3:
                type: cmd
                cmd: [/usr/bin/bash, /etc/kvmd/scripts/switch_input.sh, 3]
            switch_input_ch4:
                type: cmd
                cmd: [/usr/bin/bash, /etc/kvmd/scripts/switch_input.sh, 4]
       scheme:
            reboot_button:
                driver: reboot
                mode: output
                switch: false
            restart_kvmd_button:
                driver: restart_kvmd
                mode: output
                switch: false
            ch1_button:
                driver: switch_input_ch1
                mode: output
                switch: false
            ch2_button:
                driver: switch_input_ch2
                mode: output
                switch: false
            ch3_button:
                driver: switch_input_ch3
                mode: output
                switch: false
            ch4_button:
                driver: switch_input_ch4
                mode: output
                switch: false
       view:
            header:
                title: Controls
            table:
                - ["#PC Input"]
                - ["ch1_button|\U0001F4BB computer 1"]
                - ["ch2_button|\U0001F4BB computer 2"]
                - ["ch3_button|\U0001F4BB computer 3"]
                - ["ch4_button|\U0001F4BB computer 4"]
                - ["#PiKVM"]
                - ["reboot_button|confirm|\U0001F503 reboot", "restart_kvmd_button|confirm|restart kvmd"]

```

mdevaev, 2024-05-01:
Hello. I did not assume that the menu should close, so everything works as intended. I think I can add some kind of option for that.

mdevaev, 2024-05-12:
Released in KVMD 4.0. Use something like `ch1_button|hide|\U0001F4BB computer 1`.

closed 2024-05-12 by mdevaev