pikvm/pikvm #1313: Mouse Pad or Bluetooth mouse left click does not register , middle click on the scroller bar for connect mouse registers as left click.

mjahangirbaig, 2024-04-27:
**Describe the bug**
On the Pikvm UI , when using safari or Firefox on iPad Pro 6th Generation (MP630LL/A) iPADOS 17.4.1 when you are using the connect mouse pad or a bluetooth mouse the left click of the mouse pad or connect bluetooth mouse does not register as a left click. 

Notices that when I did a middle click on the scroll bar of the connected mouse a left click was registered on PiKVM screen.

**To Reproduce**
Steps to reproduce the behavior, like:
1. Go to 'PIKVM UI connect to a machine on Safari or Firefox on iPad Pro iPADOS 17.4.1'
2. Use a Bluetooth mouse pad or any bluetooth connect mouse
3. Do a left click anywhere on the screen the left click does not register , instead if you do a scroll bar middle click a left click is registered

**Expected behavior**
When a left click is made on the iPad it should register as a left click

**Screenshots**
[image]
[image]

**Desktop (please complete the following information):**
 - OS: iPADOS 17.4.1
 - Browser  safari or Mozilla Firefox for IPAD
 - Version : Firefox 125.2 / Safari version bundled with iPAD OS 17.4.1

**PiKVM info:**
  - Raspberry Pi board version :RPi 4 using HAT V3
  - PiKVM platform : RPi 4 HAT v3 BOX Preactivated
  - Video capture type : 1080p CSI2 with audio
  - KVMD version:  kvmd 3.329-1, kvmd-fan 0.30-1,  kvmd-oled 0.27-1, kvmd-platform-v3-hdmi-rpi4 3.329-1, kvmd-webterm 0.48-1
  - uStreamer version: `6.9-1`
  - Linux kernel: `Linux pikvm 6.6.21-4-rpi #1 SMP Wed Mar 13 20:36:42 UTC 2024 armv7l GNU/Linux`

**Additional context**
Add any other context about the problem here.

mdevaev, 2024-05-01:
I am informed that there is no such problem on 17.2, so it may be a bug in Safari. Also, can you turn off the keyboard to check if the problem persists? Leave only the mouse.

mdevaev, 2024-05-01:
On iOS Firefox (like any other browser) uses Safari engine, so the bug will persist on all browsers.

arch1mede, 2024-05-06:
I attempted to repo this and wasn't able to. I tried this on 17.2 and 17.4. In both instances mouse was working with the left click using the v3 HAT.

dericksantillan, 2024-05-11:
Same happens to me if I use my iPhone for pikvm with a mouse connected. Middle mouse registers as a left click

arch1mede, 2024-05-11:
I am unable to reproduce this issue on an iPhone or iPad, I would suggest activating VNC and using a vnc client like bVNC or Jump as I feel this offers a better user experience for mobile use. There is talk for a web revamp but I am unable to provide an ETA at this time, perhaps at that time, this will resolve the issue.

mdevaev, 2024-05-12:
I've tested in on iPad Pro. Left and right mouse buttons are working, but the middle button acts as the left one. 

I found this: https://discussions.apple.com/thread/254458713

It looks like this is some kind of fundamental problem with iOS or the Safari engine. It interprets any additional mouse button except the right one as the left one, this also works for the side buttons. I'm sorry, but I can't fix it in any way, it's beyond the capabilities of client-side JS.

closed 2024-05-12 by mdevaev

dericksantillan, 2024-05-12:
Hi Max,Just curious if you were able to replicate using a iPhone? My specific use case for mouse issues was when I use left mouse clicks with an iPhone. Thanks

mdevaev, 2024-05-12:
I didn't try, but iPad and iPhone uses the pretty same browser engine. This can't be fixed either, because we process clicks the way the browser tells us to. That is, the Web UI does not work with the mouse directly.

dericksantillan, 2024-05-12:
I see. Appreciate the troubleshooting. Thanks