pikvm/pikvm #1315: Compiling fails / problem wit pacman

etkaar, 2024-04-30:
When I try to compile PiKVM (v2-hdmi for RPI4) on Debian 11 Bullseye with `make os`, I get:

```
:: File /var/cache/pacman/pkg/glibc-2.39-2-x86_64.pkg.tar.zst is corrupted (invalid or corrupted package (PGP signature)).
Do you want to delete it? [Y/n] error: glibc: signature from "Frederik Schwan <frederik.schwan@linux.com>" is unknown trust
error: failed to commit transaction (invalid or corrupted package)

Errors occurred, no packages were upgraded.
```

mdevaev, 2024-04-30:
Please use a ready-made image for this case: https://docs.pikvm.org/flashing_os/

Sometimes the features of the host system fail during assembly.

etkaar, 2024-04-30:
Thanks. I tried it later with a new Ubuntu VPS and then it worked.

closed 2024-04-30 by etkaar