pikvm/pikvm #1316: When I click Maximize in PiKVM I expect the window to resize as I resize my browser, but it does not stay maximized

Majestic7979, 2024-05-02:
**Describe the bug**
When I click the Maximize button:
[image]
the KVM video will fill the container in the browser window... But if I drag my browser window from the corner to increase the size, the video does not increase with it. Maximize should always adjust the window to fill the screen, and if I drag the browser to make it smaller this works fine on the PiKVM, it shrinks the window to make it smaller and fit the viewport. But it does not enlarge the video to fit the viewport if I make the browser window bigger. Please fix this, I believe it is a bug because it works as expect when the window is being shrunk but does not work when it is being expanded.

arch1mede, 2024-05-06:
Hi, can you join [Discord](https://discord.gg/bpmXfz5) to discuss this? I am not able to repo this issue. I can maximize the window and make it smaller or larger. Perhaps there is some misunderstanding.

Majestic7979, 2024-05-06:
My sincerest apologies for the inconvenience, I do not have a Discord, it's not that I am uncooperative, it's because I suffered harassment there so I decided I no longer use the service. I hope you can have empathy and understand. Could you tell me what is unclear so I can explain? There is the maximize button for the browser window, and the maximize button for the kvm virtual "monitor"/vieport. I am talking about clicking the virtual monitor maximize and then resizing the browser window itself. When you resize the browser window after maximizing the pikvm virtual monitor/viewport you will see that the viewport size does not adjust to fill the window. You have to click maximize on the viewport again to fill the browser window again after each time you adjust the browser window to enlarge it. But if you shrink the browser window then the virtual monitor/viewport correctly becomes smaller/constrained to the browser window. So making the browser smaller the viewport follows the browser window size and adjusts itself. But not when you make the browser window larger, then you will see the viewport/virtual monitor will become a floating window over the gray background and you have to click maximize again to fill the browser window again.

arch1mede, 2024-05-11:
> My sincerest apologies for the inconvenience, I do not have a Discord, it's not that I am uncooperative, it's because I suffered harassment there so I decided I no longer use the service. I hope you can have empathy and understand. Could you tell me what is unclear so I can explain?
I understand, I am still unable to reproduce this issue. I would like to request that you join our Discord server to discuss this matter further. You are not required to install the Discord app; you can use the web version for a one-time connection. Github is not as actively monitored as Discord for support, so I would appreciate your joining Discord to continue this discussion.

> There is the maximize button for the browser window, and the maximize button for the kvm virtual "monitor"/vieport. I am talking about clicking the virtual monitor maximize and then resizing the [...]
Either this is being explained the same or I am not able to reproduce this issue, I have tried again hitting the same button as was mentioned in the Original Posting, I am able to manipulate the window without issue.

I would rather we discuss this further on Discord, you can either join the channel or DM me directly to discuss this further but this seems more like a support issue than a PiKVM issue.

mdevaev, 2024-05-11:
In fact, it works as intended. We don't have an automatic recycle in many cases. I would not like to implement this because it would greatly complicate the heuristics of window operations.

mdevaev, 2025-05-14:
Time to fix it.

mdevaev, 2025-05-18:
Released in 4.72, please update with `pikvm-update`.

closed 2025-05-18 by mdevaev

Majestic7979, 2025-05-22:
Dude you always blow my mind!!!!!!!!!! Never stop improving your product.. Amazing! Thanks so much! If you are able to somehow give us the remote webcam (pass the webcam stream from the device I am using to control the PiKVM from my browser, to the PiKVM so the software running on it thinks the webcam is plugged into USB or something else) that would be incredible! Thanks for your continual improvement mindset!

mdevaev, 2025-05-22:
You're welcome. There is and issue about webcams as I remember, maybe I will add it in the future.

Majestic7979, 2025-06-04:
@mdevaev Maxim, is it possible to start releasing "release notes" please? So that I can keep track of what each update is doing? That would help understand new additions, removals, amendments to the system, etc. Thanks a lot!