pikvm/pikvm #1317: PiKVM V4 Mini - No power after EDID reconfiguration

va7acq, 2024-05-03:
I bought a new PiKVM V4 Mini from pishop.ca last month, and I've been using it for almost a week to remotely control a HP 845 G7 laptop. It worked nearly perfectly, except it insisted on capturing in 1920x1200 even though the laptop was putting out 1920x1080. Setting the laptop to 1920x1200 worked, but the resulting video was slightly too tall for my monitor & browser leading to a small amount of rescaling & distortion. I use Chrome on Fedora 40.

I found this page in the documentation on how to disable the 1920x1200 mode: https://docs.pikvm.org/edid/#disable-1920x1200-on-pikvm-v4

I followed the instructions carefully, and selected the v4mini preset:

```
# rw
# kvmd-edidconf --import-preset=v4mini.no-1920x1200
# reboot
```

The kvmd-edidconf command produced some brief output, but there was no error message so I continued with the reboot. Instead, the PiKVM powered off, including the red 5V LED. 

Now it seems to be almost bricked. The green "SR" LED comes on beside the HDMI connector when I connect the video to my laptop, and a green LED comes on when I plug a USB-C cable into the IOIOI port. But, the red 5V LED never comes on, and there is no response on the serial port. There is nothing displayed on the OLED screen.

I tested the PiKVM's power supply with a Pi 400 and it seems to work fine. I also tried the Pi 400 power supply on the PiKVM, but had the same results with no power. I let it sit overnight and tried again, but it still behaves the same.

I had already upgraded it to PiKVMD v3.333 before the failure, and that seemed to go smoothly.

Is there anything I can do to "reset" the PiKVM or bring it back to life? 

Thanks! 

Josh

mdevaev, 2024-05-04:
Hi. This cannot be related, because changing the EDID does not change anything in the hardware, but only the contents of single file. Perhaps the problem was caused by other actions. Please reflash the memory card and try again: https://docs.pikvm.org/flashing_os

va7acq, 2024-05-05:
Thanks for the reply! I re-flashed the memory card as suggested, but there was no change in behavior. 

I assume the red 5V LED should come on as soon as the power supply is connected. Is it actually controlled by software? It seemed to work perfectly until I rebooted it.

Cheers,

Josh

mdevaev, 2024-05-05:
The red led is activity led. What about the green frontal led? Is it on?

arch1mede, 2024-05-05:
You can open a ticket with support@pishop.ca and discuss there, some information would be needed to proceed. Its possible you might have a CM4 failure.

mdevaev, 2024-05-07:
Okay, I'm closing this. If you want to replace a faulty device, please write to the above email.

closed 2024-05-07 by mdevaev