pikvm/pikvm #1322: As user, I need a viewport of the target at a resolution of 1:1

msdobrescu, 2024-05-09:
**Is your feature request related to a problem? Please describe.**
I have observed a better image quality of the image when the target resolution is scaled at 100% in the webrtc app than at less than 100%.

**Describe the solution you'd like**
I'd like to have the native target resolution _locked_ in the webrtc view even though the web page is smaller than the needed resolution to display the full target desktop, but if that happens, scrollbars to be shown to access and change the visible area.

**Describe alternatives you've considered**
TigerVNC has this kind of approach.

**Additional context**
As I have observed, it is possible that the page elements style and javascript can be tuned for the case.

mdevaev, 2024-05-10:
TigerVNC uses this approach due to the limitations of the GUI library used. I don't think I'll be doing a similar function. If you need better quality, please try fullscreen mode.

msdobrescu, 2024-05-10:
Well, the quality is not about how nice some can get the image, but readability in certain circumstances, so bigger image can be used to better analyze some part of the screen's target in some cases when that is needed. Think of it as a being a magnifier. It is not a true magnifier as it shows at the image at 100% instead of scaled version at 50%-75%, but it is a way to actually see at pixel without other blurring or similar effect due to scaling to a smaller size.

mdevaev, 2025-11-26:
I'm thinking about your request, it also fits in with requests to improve the mobile interface, so I've combined these tasks. Please subscribe to the new issue to keep an eye on it: https://github.com/pikvm/pikvm/issues/1614

closed 2025-11-26 by mdevaev