pikvm/pikvm #1324: failure to update (dependency issues) across several PiKVM units

Wilford888, 2024-05-13:
**Describe the bug**
pacman -Syu fails across devices that have been upgraded in the recent past with the following error (answering Y or N makes no difference, error is the same):
:: Replace dbus-python with extra/python-dbus? [Y/n] n
resolving dependencies...
looking for conflicting packages...
error: failed to prepare transaction (could not satisfy dependencies)
:: unable to satisfy dependency 'python<3.12' required by kvmd
:: installing python (3.12.3-1) breaks dependency 'python<3.12' required by python-luma-core
:: installing python (3.12.3-1) breaks dependency 'python<3.12' required by python-luma-oled
:: installing python (3.12.3-1) breaks dependency 'python<3.12' required by python-pyghmi
:: installing python (3.12.3-1) breaks dependency 'python<3.12' required by python-pyrad
:: installing python (3.12.3-1) breaks dependency 'python<3.12' required by python-pyserial-asyncio
:: installing python (3.12.3-1) breaks dependency 'python<3.12' required by python-raspberry-gpio
:: installing python (3.12.3-1) breaks dependency 'python<3.12' required by python-smbus2
:: installing python (3.12.3-1) breaks dependency 'python<3.12' required by python-spidev

**To Reproduce**
Steps to reproduce the behavior, like:
1. run pacman -Syu

**Expected behavior**
Successful update 

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS:  Windows Terminal program
 - Browser LibreWolf

**PiKVM info:**
  - Raspberry Pi board version 3
  - PiKVM platform v3]
  - KVMD version: `pacman -Q | grep kvmd` generates:
kvmd 3.333-1
kvmd-fan 0.30-1
kvmd-oled 0.27-1
kvmd-platform-v3-hdmi-rpi4 3.333-1
kvmd-webterm 0.50-1

  - uStreamer version: `pacman -Q | grep ustreamer` generates:
ustreamer 6.11-1

  - Linux kernel: `uname -a`
Linux PiKVM-214 6.6.21-4-rpi #1 SMP Wed Mar 13 20:36:42 UTC 2024 armv7l GNU/Linux

**Additional context**
None

rheindevde, 2024-05-13:
same here:

pacman -Syu
:: Synchronizing package databases...
 core is up to date
 extra is up to date
 community is up to date
 alarm is up to date
 aur is up to date
 pikvm is up to date
:: Starting full system upgrade...
:: Replace dbus-python with extra/python-dbus? [Y/n] Y
resolving dependencies...
looking for conflicting packages...
error: failed to prepare transaction (could not satisfy dependencies)
:: unable to satisfy dependency 'python<3.12' required by kvmd
:: installing python (3.12.3-1) breaks dependency 'python<3.12' required by python-luma-core
:: installing python (3.12.3-1) breaks dependency 'python<3.12' required by python-luma-oled
:: installing python (3.12.3-1) breaks dependency 'python<3.12' required by python-pyghmi
:: installing python (3.12.3-1) breaks dependency 'python<3.12' required by python-pyrad
:: installing python (3.12.3-1) breaks dependency 'python<3.12' required by python-pyserial-asyncio
:: installing python (3.12.3-1) breaks dependency 'python<3.12' required by python-raspberry-gpio
:: installing python (3.12.3-1) breaks dependency 'python<3.12' required by python-smbus2
:: installing python (3.12.3-1) breaks dependency 'python<3.12' required by python-spidev

e-hollebone, 2024-05-13:
comment to follow issue, I have the same problem

DialectKBA-Teodor, 2024-05-13:
Same issue here, but with a few differences to the original report:

- Raspberry Pi board version 4

- PiKVM platform v2

- KVMD version: `pacman -Q | grep` kvmd generates:
kvmd 3.332-1
kvmd-fan 0.30-1
kvmd-oled 0.27-1
kvmd-platform-v2-hdmiusb-rpi4 3.332-1
kvmd-webterm 0.49-1

4plb0mb, 2024-05-13:
Same here, except for a couple things:

- Hardware are 2x pikvm v4 plus

- update command is pikvm-update (although I know that it's "just" a script that also uses pacman)

Both v4's `pacman -Q | grep kvmd`:
kvmd 3.333-1
kvmd-fan 0.30-1
kvmd-oled 0.27-1
kvmd-platform-v4plus-hdmi-rpi4 3.333-1
kvmd-webterm 0.50-1

and `pacman -Q | grep ustreamer`
ustreamer 6.11-1

mdevaev, 2024-05-13:
This happens when Python is updated in the distribution. We have a pinned version, so that nothing breaks because of the update. Everything worked as it should.

Updated now.

closed 2024-05-13 by mdevaev

dougmaitelli, 2024-05-13:
@mdevaev 

Thanks for the quick follow-up, after your comment I see now that the error changed, none of the original dependency conflicts are there, but there is one new one:

```
error: failed to prepare transaction (could not satisfy dependencies)
:: installing python (3.12.3-1) breaks dependency 'python<3.12' required by python-ajsonrpc
```

Wilford888, 2024-05-17:
Thank you.  Issue resolved.

tim-lux, 2024-05-25:
I also ran into this issue:
`removing python-ajsonrpc breaks dependency 'python-ajsonrpc' required by platformio`
(following a chain of dependencies, so removing it did not

As some suggested on the discord, reinstalling "python-ajson" worked for me (rpi 4, pikvm 3): `**pacman -S python-ajsonrpc**`