pikvm/pikvm #1325: Burnt colors for h.264

premekj, 2024-05-13:
**Describe the bug**
Colors of the h.264 stream looks too bright.

**To Reproduce**
clean installation (i have also tried some modifications in override.yaml but without success)
latest RPI4 with CSI bridge
H.264 stream has more brighter colors than MJPEG and real picture on monitor, tested web interface and vnc

**Expected behavior**
Same colors for h.264 as well

**Screenshots**
[image]
for example, see the difference around mouse pointer - background of the button
 "Add Calendar" blends in with the surroundings. Or "Sun 12" column is white, not light gray. 

**Desktop (please complete the following information):**
 - OS: Windows 10 (and tested on Win11)
 - Browser: latest chrome & firefox 
 - VNC client: tested TigerVNC (it used h.264) with same symptoms, TurboVNC (tight encoding) is correct
 - local gigabit network without firewall

**PiKVM info:**
  - RPi 4
  - v2-hdmi
  - CSI bridge (tried two different)
  - KVMD version: 4.2-1
  - uStreamer version: 6.11-2
  - Linux kernel: Linux pikvm 6.6.21-4-rpi # 1 SMP Wed Mar 13 20:36:42 UTC 2024 armv7l GNU/Linux

mdevaev, 2024-05-13:
Try to update EDID: `kvmd-edidconf --import-preset=v2` and reboot.

premekj, 2024-05-13:
```
[root@pikvm ~]# rw
+ mount -o remount,rw /
+ mount -o remount,rw /boot
+ set +x
=== PiKVM is in Read-Write mode ===
[root@pikvm ~]# kvmd-edidconf --import-preset=v2
Manufacturer ID: LNX
Product ID:      0x7770 (30576)
Serial number:   0x01010101 (16843009)
Monitor name:    PiKVM
Monitor serial:  CAFEBABE
Audio:           no
[root@pikvm ~]# reboot
```
done, but without any change in picture

mdevaev, 2024-05-13:
What changes did you make in configs and image?

The colors should be good unless you've tried using hdmi passthrough.

premekj, 2024-05-20:
In the past few days I have some time to experiment with it. The problem occours only with signal from Dell docking station (TB16), used setup: notebook -> (usb-c cable) -> dock Dell TB16 -> (hdmi cable) -> CSI bridge. I have tested several different notebooks (Dell, HP and one HP desktop) with this setup, the colors were never right. 

But with direct connections notebook (or any other hdmi source) -> (hdmi) -> CSI bridge working well with H.264

I have tried on clean PiKVM image:
* change hdmi cable, csi bridge
* switch resolution and refresh rate on notebook(s)
* change EDID
* H.264 kbps & gop

Is it possible to force TigerVNC viewer to use Tight compression? I have tried to set Tight compression for TigetVNC, but it always use H.264. Do you think, is it a bug in TigerVNC?

mdevaev, 2024-05-20:
TigerVNC is always prefering H.264 because it provides better compression.

I suppose the key problem is USB-C converter.

premekj, 2024-05-21:
I updated the firmware in dock to the latest, but the problem still persist.
I will try another docks in the next few days and I'll let you know.

I'm currently using TigerVNC 1.12.0 (it lacks h.264 support) and it is ok. Is it possible to turn off H.264 support for VNC on PiKVM in the settings?

mdevaev, 2024-05-22:
Add this to `/etc/kvmd/override.yaml` to disable H.264:
```yaml
vnc:
    memsink:
        h264:
            sink: ""
```

mdevaev, 2024-05-26:
There is a small chance that your host has cached the EDID data. You can try some other EDID from existing monitors that you haven't tried yet (http://docs.pikvm.org/edid)

mdevaev, 2024-07-24:
Answered I guess.

closed 2024-07-24 by mdevaev

sbytnar, 2025-07-11:
I have a similar situation with H.264.
Washed out: notebook -> (USB 3.0 cable) -> Dell D3100 dock -> (hdmi cable) -> PiKVM v3
Not washed out: notebook -> (hdmi cable) -> PiKVM v3.
EDID changes didn't matter. I gave up with the first and am using the second configuration.

mdevaev, 2025-07-15:
@sbytnar This is most likely due to the conversion of color spaces, if your dock is too smart and converts the signal. I don't think there's anything I can do here, sorry.