pikvm/pikvm #1326: Make the audio on h264 always start at 100%

Majestic7979, 2024-05-17:
When I start my computer with audio, it never starts at 0% and makes me change it to 100% every time. So why should PiKVM be the same? It should remember my last setting and stay there.

Every morning when I start my PiKVM I have to drag the volume slider from 0 to 100, it's rather inconvenient. Can this please be fixed? The slider should remember its last position so if I set to 100 it should remain there, and if I set to 0 it should remain at 0 but it should not keep resetting itself to 0 every time I open a browser window. The default should be 100.

mdevaev, 2024-05-17:
Duplicate of #1201

This is currently not implemented due to browser limitations.

Majestic7979, 2024-05-21:
Isn't it possible to set a cookie? The instructions could tell the user to go into the page settings and "allow audio" to "always" in case of Chrome... Then the cookie would take care of setting the volume to 100? I don't know the way it works so just some questions to think about in case you haven't :)

mdevaev, 2024-05-21:
The problem is described in the link above ^^^ It won't work just like that, even with magic cookies.