pikvm/pikvm #1329: New plugin for 52pi hat

mmoerz, 2024-05-20:
I tried to write a new plugin for a relay hat (https://wiki.52pi.com/index.php/EP-0099).

`# ========================================================================== #
#                                                                            #
#    KVMD - The main PiKVM daemon.                                           #
#                                                                            #
#    Copyright (C) 2018-2024  Maxim Devaev <mdevaev@gmail.com>               #
#                                                                            #
#    Plugin for 52pi hat                                                     #
#    Copyright (C) 2024 Michael Moerz <michael@moerz.co.at>                  #
#                                                                            #
#    This program is free software: you can redistribute it and/or modify    #
#    it under the terms of the GNU General Public License as published by    #
#    the Free Software Foundation, either version 3 of the License, or       #
#    (at your option) any later version.                                     #
#                                                                            #
#    This program is distributed in the hope that it will be useful,         #
#    but WITHOUT ANY WARRANTY; without even the implied warranty of          #
#    MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the           #
#    GNU General Public License for more details.                            #
#                                                                            #
#    You should have received a copy of the GNU General Public License       #
#    along with this program.  If not, see <https://www.gnu.org/licenses/>.  #
#                                                                            #
# ========================================================================== #

import asyncio
import contextlib
import functools

from typing import Callable
from typing import Any

# replaced with 52pi i2c stuff
import time
import smbus

from ...logging import get_logger

from ... import tools
from ... import aiotools

from ...yamlconf import Option

from ...validators.basic import valid_number

from . import GpioDriverOfflineError
from . import UserGpioModes
from . import BaseUserGpioDriver

# =====
class Plugin(BaseUserGpioDriver):
    # https://wiki.52pi.com/index.php/EP-0099

    def __init__(  # pylint: disable=super-init-not-called
        self,
        instance_name: str,
        notifier: aiotools.AioNotifier,

        device_bus: int,
        device_addr: int,
    ) -> None:

        super().__init__(instance_name, notifier)

        self.__device_bus = device_bus
        self.__device_addr = device_addr

        self.__device:(smbus.SMBus | None) = None  # type: ignore
        self.__stop = False
[... 150 lines cut ...]
                - ["#Reset"]
                - []
`

The expected behavior would be that the switches in the webgui trigger the relays.
However what happens is:
webgui does display an additional menu item "switches" with the two buttons Up and down.
I can toggle or click the buttons, and no relays are trigger.
In journal with `journalctl -f -u kvmd` I can see "click":

`
May 12 01:02:33 pikvm kvmd[19966]: kvmd.apps.kvmd.ugpio              INFO --- Pulsed Output(kvmUp, driver=GPIO(p52pi_one), pin=1) with delay=0.10
`
and toggle:
`
May 12 01:03:41 pikvm kvmd[19966]: kvmd.apps.kvmd.ugpio              INFO --- Ensured switch Output(kvmUp, driver=GPIO(p52pi_one), pin=1) to state=1
May 12 01:03:41 pikvm kvmd[19966]: aiohttp.access                    INFO --- [admin (token) / 10.10.100.148] 'POST /gpio/switch?channel=kvmUp&state=1 HTTP/1.0' => 200; size=194 --- referer='https://10.2.200.41/kvm/'; user_agent='Mozilla/5.0 (X11; Linux x86_64; rv:124.0) Gecko/20100101 Firefox/124.0'
May 12 01:03:43 pikvm kvmd[19966]: kvmd.apps.kvmd.ugpio              INFO --- Ensured switch Output(kvmUp, driver=GPIO(p52pi_one), pin=1) to state=0
May 12 01:03:43 pikvm kvmd[19966]: aiohttp.access                    INFO --- [admin (token) / 10.10.100.148] 'POST /gpio/switch?channel=kvmUp&state=0 HTTP/1.0' => 200; size=194 --- referer='https://10.2.200.41/kvm/'; user_agent='Mozilla/5.0 (X11; Linux x86_64; rv:124.0) Gecko/20100101 Firefox/124.0'
`

How do I make the logging inside the module output it's messages?
the plugin does not seam to log to journald.
I suspect I must have broken something regarding the handling of the module, because if I use the object in a script of my own, it flips the relays.
Additionally after starting kvmd with my plugin the i2c bus sort of hangs - I am unable to flip any relays on the hat after starting kvmd. the only way to fix this is to reboot the pi.

mmoerz, 2025-04-22:
yeah, I still like to make this plugin work.
however it's not urgent, since i created a workaround using the cmd plugin and a script to set the pins. that works fine so far, but I think it's an 'ugly' solution and a bit cumbersome to setup.
I'm going to update my pikvm with the current version and see what that brings me

mdevaev, 2025-05-03:
The only problem I see is blocking smbus device, it should be wrapped for asyncio to avoid global lock.

mdevaev, 2025-11-09:
I'm going to refine it and take it upstream. Are you still interested? :)

mmoerz, 2025-11-16:
yeah, I can give it a spin, or provide logs/debug when necessary

mdevaev, 2025-12-04:
@mmoerz Sup?

mdevaev, 2026-02-24:
@mmoerz Hello?