pikvm/pikvm #1330: Update not working

mcnahum, 2024-05-21:
**Describe the bug**

The goal is simply to update the device

**To Reproduce**

curl https://files.pikvm.org/update-os.sh | bash

  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  2948  100  2948    0     0  17816      0 --:--:-- --:--:-- --:--:-- 17866
++ whoami
+ '[' root '!=' root ']'
+ rw
+ mount -o remount,rw /
+ mount -o remount,rw /boot
+ set +x
=== PiKVM is in Read-Write mode ===
+ pacman -Syy
:: Synchronizing package databases...
 core                                                                                                   239.4 KiB   452 KiB/s 00:01 [################################################################################] 100%
 extra                                                                                                    9.0 MiB  4.32 MiB/s 00:02 [################################################################################] 100%
 community                                                                                               45.0   B   562   B/s 00:00 [################################################################################] 100%
 alarm                                                                                                   94.8 KiB  1354 KiB/s 00:00 [################################################################################] 100%
 aur                                                                                                      9.3 KiB   117 KiB/s 00:00 [################################################################################] 100%
 pikvm                                                                                                   10.6 KiB  76.0 KiB/s 00:00 [################################################################################] 100%
++ pacman -S -u --print-format %n
++ wc -l
error: failed to prepare transaction (could not satisfy dependencies)
+ '[' 2 -eq 0 ']'
+ grep -q '^C.UTF-8 UTF-8' /etc/locale.gen
+ for _pkg in rpi-eeprom rpi4-eeprom
+ pacman -Q rpi-eeprom
+ for _pkg in rpi-eeprom rpi4-eeprom
+ pacman -Q rpi4-eeprom
+ _yes='--noconfirm --ask=4 --overwrite \*'
+ pacman -Q raspberrypi-utils libgpiod
+ pacman --noconfirm --ask=4 --overwrite '\*' -Su
:: Starting full system upgrade...
:: Replace dbus-python with extra/python-dbus? [Y/n] 
:: Replace raspberrypi-bootloader-x-pikvm with pikvm/raspberrypi-bootloader-pikvm? [Y/n] 
resolving dependencies...
:: There are 2 providers available for dbus-units:
:: Repository core
   1) dbus-broker-units  2) dbus-daemon-units

Enter a number (default=1): 
looking for conflicting packages...
:: kvmd and platformio are in conflict. Remove platformio? [y/N] 
:: kvmd and avrdude-pikvm are in conflict. Remove avrdude-pikvm? [y/N] 
error: failed to prepare transaction (could not satisfy dependencies)
:: installing python (3.12.3-1) breaks dependency 'python<3.12' required by python-ajsonrpc

**PiKVM info:**

Board 4

[root@pikvm /]# pacman -Q | grep kvmd
kvmd 3.257-2
kvmd-fan 0.30-1
kvmd-oled 0.25-1
kvmd-platform-v4plus-hdmi-rpi4 3.257-2
kvmd-webterm 0.43-1

[root@pikvm /]# pacman -Q | grep ustreamer
ustreamer 6.12-1

[root@pikvm /]# uname -a
Linux pikvm 5.15.92-1-rpi-ARCH #2 SMP Sat Sep 9 00:58:18 MSK 2023 armv7l GNU/Linux

mdevaev, 2024-05-21:
Please re-run it and follow instructions.

mcnahum, 2024-05-21:
I'm  impressed! 
Everything worked perfectly, and I noticed you already updated the FAQ. 
Respect!

closed 2024-05-22 by mcnahum