pikvm/pikvm #1332: Microsoft VNC client on MacOS doesn't allow to enter admin/admin credentials for PiKVM

yurivict, 2024-05-23:
**Describe the bug**
Remmina is able to connect to PiKVM fine from FreeBSD using admin/admin credentials.
However, the same fails from the Microsoft VNC client on MacOS.
It displays the login screen, but login name is grayed out, and only password can be entered.
This prevents connection from MacOS.

Is there a workaround for this problem?

mdevaev, 2024-05-23:
I believe this client is somehow incorrectly or not fully implementing [password authorization](https://github.com/rfbproto/rfbproto/blob/master/rfbproto.rst#vencrypt). I usually recommend using TigerVNC in such cases.

mdevaev, 2024-05-26:
Is it? https://apps.microsoft.com/detail/9pj2zqlfkwws?hl=en-mm&gl=MM

yurivict, 2024-05-26:
* Correction:
It is RealVNC for MacOS (not MS VNC)

Sorry for the confusion.

mdevaev, 2024-05-27:
Ah, RealVNC. Paradoxically, RealVNC is very far from open VNCs and for the most part implements its own proprietary protocols. It has never worked at all and I doubt it will.

yurivict, 2024-05-27:
Ok, thank you.

closed 2024-05-27 by yurivict