pikvm/pikvm #1333: kvmd-fan.service unset environment variable: KVMD_FAN_ARGS

indirection, 2024-05-23:
**Describe the bug**
Fan is working without issue, however, PiKVM is reporting Fan Error.

**To Reproduce**
1. Power-on system
2. Login to PiKVM
3. See error message in top left corner.

**Expected behavior**
No error. There seems to be an issue with the referenced file: libiniparser.so. (see log below)

**Screenshots**
[image]

**Desktop (please complete the following information):**

- OS: macOS, Linux, Windows 11
- Browser Firefox, Chrome, Safari
- Version current releases
- VNC client N/A

**PiKVM info:**

- Raspberry Pi board version CM4 Rev 1.1
- PiKVM platform PiKVM v4 Plus
- Video capture type ?
- KVMD version: `pacman -Q | grep kvmd`

```bash
[root@pikvm kvmd]# pacman -Q | grep kvmd
kvmd 4.2-1
kvmd-fan 0.30-1
kvmd-oled 0.27-1
kvmd-platform-v4plus-hdmi-rpi4 4.2-1
kvmd-webterm 0.50-1
```

- uStreamer version: `pacman -Q | grep ustreamer`

```bash
[root@pikvm kvmd]# pacman -Q | grep ustreamer
ustreamer 6.12-1
```

- Linux kernel: `uname -a`

```bash
[root@pikvm kvmd]# uname -a
Linux pikvm 6.6.21-4-rpi #1 SMP Wed Mar 13 20:36:42 UTC 2024 armv7l GNU/Linux
```

- Platform:
  - Base: Raspberry Pi Compute Module 4 Rev 1.1
- Temperature:
  - CPU: 36.024°C
- Throttling:
  - Freq capped: No
  - Throttled: No
  - Undervoltage: No
- Fan:
  - Status: **Not available**
- KVMD: 4.2
  - Streamer: 6.12 (ustreamer)
    - HAS_PDEATHSIG: Yes
    - WITH_GPIO: Yes
    - WITH_PTHREAD_NP: Yes
    - WITH_SETPROCTITLE: Yes
    - WITH_SYSTEMD: Yes
  - Linux kernel:
    - Machine: armv7l
    - Release: 6.6.21-4-rpi
    - Version: 1 SMP Wed Mar 13 20:36:42 UTC 2024

**Additional context**

Log:

```log
[2024-05-23 04:10:10 kvmd-fan.service] --- kvmd-fan.service: Referenced but unset environment variable evaluates to an empty string: KVMD_FAN_ARGS
[2024-05-23 04:10:10 kvmd-fan.service] --- /usr/bin/kvmd-fan: error while loading shared libraries: libiniparser.so.1: cannot open shared object file: No such file or directory
[2024-05-23 04:10:12 kvmd.service] --- kvmd.apps.kvmd.info.fan ERROR --- Can't read fan state: Cannot connect to unix socket /run/kvmd/fan.sock ssl:default [No such file or directory]
```

```bash
File output:
[root@pikvm kvmd]# grep -ri "KVMD_FAN_ARGS" / 2>/dev/null
/usr/lib/systemd/system/kvmd-fan.service:ExecStart=/usr/bin/kvmd-fan --config=?/etc/kvmd/fan.ini $KVMD_FAN_ARGS
[root@pikvm kvmd]# cat /etc/kvmd/fan.ini
[main]
pwm_pin = 13
hall_pin = 16

[speed]
idle = 27
low = 27

[server]
unix = /run/kvmd/fan.sock
unix_rm = 1
unix_mode = 666
[root@pikvm kvmd]#
```

adammerkley, 2024-05-23:
Also seeing this issue after running pikvm-update tonight

mdevaev, 2024-05-23:
Please run `pikvm-update` again. Fixed now.

closed 2024-05-23 by mdevaev