pikvm/pikvm #1334: Prevent sending repetitive keypress sequence when there is a lag or slowdown

plia7, 2024-05-24:
**Describe the bug**
When there is a lag or slowdown sending a keystroke can result in repetitive keypress sequence.

**To Reproduce**
1. Try to type some words in a text editor in the host during a lag or slowdown:

```
hello world
```

**Actual behavior**
It ends up sending repetitive keypress sequence on some letters:

```
helllllllooooo wwwwwworrrrrrrlddd
```

**Expected behavior**
No character repeats if you only sent each character once despite any network lag or slowdown/interruption:

```
hello world
```

**Additional context**
Try to simulate the conditions for a network slowdown or interruption to replicate the issue.
If this can't be as part of standard behavior, please consider adding a setting or a special mode that can be prevent this behavior.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
Windows 11
 - Browser [e.g. chrome, safari]
Google Chrome Version 125.0.6422.77 (Official Build) (64-bit)
 - Version [e.g. 22]
 - VNC client (if used)

**PiKVM info:**
  - Raspberry Pi board version [e.g. RPi 4]
PiKVM v3 Pre-Assembled
  - PiKVM platform [e.g. v2-hdmi]
  - Video capture type [e.g. CSI bridge]
  - KVMD version: `pacman -Q | grep kvmd`
kvmd 3.321-1
kvmd-fan 0.30-1
kvmd-oled 0.26-1
kvmd-platform-v3-hdmi-rpi4 3.321-1
kvmd-webterm 0.48-1
  - uStreamer version: `pacman -Q | grep ustreamer`
ustreamer 6.4-1
  - Linux kernel: `uname -a`
Linux pikvm 6.6.21-4-rpi # 1 SMP Wed Mar 13 20:36:42 UTC 2024 armv7l GNU/Linux

Thanks.

plia7, 2024-06-03:
@mdevaev Sup? Any updates?

mdevaev, 2024-06-04:
I haven't come up with a good solution yet.

plia7, 2024-06-04:
> I haven't come up with a good solution yet.

How about adding a mode in keyboard.js
for it to auto send keyup for letters and numbers?

Unless you can send keystrokes without key down first?

Thanks.

cRoCx, 2024-06-06:
I'm having the same issue. Happens irregularly when latency is high and/or packet loss occurs, exactly as @plia7 described it above.
In these cases editing text is time consuming. My workaround is removing the repeated characters by selecting them with the mouse and then press Ctrl + x.

plia7, 2024-06-10:
@mdevaev @cRoCx  Here is a solution that might work, in /usr/share/kvmd/web/share/js/kvm/keyboard.js
you want to define a variable:

```
   var __keypad = null;

    **var pressedKeys = new Set();**
```

Then make sure you clear it when needed:

```
   self.releaseAll = function() {
        __keypad.releaseAll();
        **pressedKeys.clear();**
    };
```

Then the actual code:

```
   var __keyboardHandler = function(event, state) {
        event.preventDefault();
        const key = event.key;

        // Check if the key is a letter, number or special character
        const isLetterNumberOrSC = /^[a-zA-Z0-9!@#$%^&*()_+\-=\[\]{};':"\\|,.<>\/?`~]$/.test(key);

        if (state) {  // Key down event
            if (!pressedKeys.has(event.code)) {
                pressedKeys.add(event.code);
                __keypad.emitByKeyEvent(event, state);

                if (isLetterNumberOrSC) {
                    // Immediately send a key up event for letters, numbers or special characters
                    setTimeout(() => __keypad.emitByKeyEvent(event, false), 0);
                    pressedKeys.delete(event.code);
                }
            }
        } else {  // Key up event
            if (pressedKeys.has(event.code)) {
                pressedKeys.delete(event.code);
                __keypad.emitByKeyEvent(event, state);
            }
        }
    };
```

Explanation: It checks to see that if the key is a letter, number or special character it will immediately send a key up, so it won't drag that character. You might adjust the set of characters used for this. I don't think white space need to be included.
This also won't break any keyboard shortcuts that typically rely on combos of alt/shift/ctrl keys.
I'll be testing it and report back, but so far, I can already see that it stop immediately if I try to drag it, so I think that's what is needed, I never need to drag these characters. White space is a different story, so I left it out.

Thanks.

dsatcom, 2024-06-10:
I have same issue as well. Would be great to have a toggle so it sends key up and key down at same time for a-Z

plia7, 2024-06-10:
> I have same issue as well. Would be great to have a toggle so it sends key up and key down at same time for a-Z

See my temporary solution for this here: https://github.com/pikvm/pikvm/issues/1334#issuecomment-2158119652

Until @mdevaev provides an official solution.

tlhonmey, 2024-09-05:
Of note:  It doesn't seem to have this problem when using the paste option in the "Text" menu.  This is, however, quite cumbersome to use.  A simple toggle in the keyboard hid options to send keyup in the same packet as keydown for everything not expected to be held down (shift, ctrl, etc.) would be a reasonable solution I would think.

mdevaev, 2024-09-09:
JFYI I'm thinking of a server-side implementation for this.

plia7, 2024-12-11:
> > I have same issue as well. Would be great to have a toggle so it sends key up and key down at same time for a-Z
> 
> See my temporary solution for this here: [#1334 (comment)](https://github.com/pikvm/pikvm/issues/1334#issuecomment-2158119652)
> 
> Until @mdevaev provides an official solution.

@mdevaev @tlhonmey 
Actually I just reproduced the issue again under certain lag/slowdown conditions even with the suggested temporary solution in place, so it might not be perfect for all cases.

cRoCx, 2024-12-14:
@plia7 Could you post some kind of diff of your workaround? I would also like to try it, but I don't know where exactly I should paste it in `/usr/share/kvmd/web/share/js/kvm/keyboard.js` .
The issue happens to me all the time, when I'm on the go on a mobile connection (train, car, ...).

plia7, 2024-12-15:
> @plia7 Could you post some kind of diff of your workaround? I would also like to try it, but I don't know where exactly I should paste it in `/usr/share/kvmd/web/share/js/kvm/keyboard.js` . The issue happens to me all the time, when I'm on the go on a mobile connection (train, car, ...).

Yes, you want to apply it inside your /usr/share/kvmd/web/share/js/kvm/keyboard.js file (you will need to ssh and use nano or and reupload the file with the changes in WinSCP - you have to be in write mode (rw) and don't forget to change back to read mode after (ro)) - This is based on the file in version 4.2.1, it could look slightly different in yours if using a higher version:

[image]

[image]

[image]

The code snippets used above:

`    var pressedKeys = new Set();
`
`    pressedKeys.clear();`

```
      const key = event.key;

      // Check if the key is a letter, number or special character
      const isLetterNumberOrSC = /^[a-zA-Z0-9!@#$%^&*()_+\-=\[\]{};':"\\|,.<>\/?`~]$/.test(key);

      if (state) {  // Key down event
          if (!pressedKeys.has(event.code)) {
              pressedKeys.add(event.code);
              __keypad.emitByKeyEvent(event, state);

              if (isLetterNumberOrSC) {
                  // Immediately send a key up event for letters, numbers or special characters
                  setTimeout(() => __keypad.emitByKeyEvent(event, false), 0);
                  pressedKeys.delete(event.code);
              }
          }
      } else {  // Key up event
          if (pressedKeys.has(event.code)) {
              pressedKeys.delete(event.code);
              __keypad.emitByKeyEvent(event, state);
          }
      }
```

And as mentioned, it doesn't seem to work for all cases, so if there is slow network condition, it might not prevent the issue from happening, but give it a try and let us know if it's any better.

deftdawg, 2024-12-27:
Here's how to patch it including @plia7's code as an  inline diff (about to to try it myself, hope it helps):
```sh
# ssh as root
rw # make system writable
cd /
cat << 'EOF' | patch -p0 --ignore-whitespace
--- /usr/share/kvmd/web/share/js/kvm/keyboard.js        2024-12-27 05:57:20.269909319 +0000
+++ /usr/share/kvmd/web/share/js/kvm/keyboard.js        2024-12-27 06:01:05.029167762 +0000
@@ -34,6 +34,8 @@
 
        var __keypad = null;
 
+    var pressedKeys = new Set();
+
        var __init__ = function() {
                __keypad = new Keypad("div#keyboard-window", __sendKey, true);
 
@@ -88,6 +90,7 @@
 
        self.releaseAll = function() {
                __keypad.releaseAll();
+        pressedKeys.clear();
        };
 
        self.emit = function(code, state) {
@@ -126,8 +129,29 @@
 
        var __keyboardHandler = function(event, state) {
                event.preventDefault();
-               __keypad.emitByKeyEvent(event, state);
-       };
+        const key = event.key;
+
+        // Check if the key is a letter, number or special character
+        const isLetterNumberOrSC = /^[a-zA-Z0-9!@#$%^&*()_+\-=\[\]{};':"\\|,.<>\/?`~]$/.test(key);
+
+        if (state) {  // Key down event
+            if (!pressedKeys.has(event.code)) {
+                pressedKeys.add(event.code);
+                __keypad.emitByKeyEvent(event, state);
+
+                if (isLetterNumberOrSC) {
+                    // Immediately send a key up event for letters, numbers or special characters
+                    setTimeout(() => __keypad.emitByKeyEvent(event, false), 0);
+                    pressedKeys.delete(event.code);
+                }
+            }
+        } else {  // Key up event
+            if (pressedKeys.has(event.code)) {
+                pressedKeys.delete(event.code);
+                __keypad.emitByKeyEvent(event, state);
+            }
+        }
+    };
 
        var __sendKey = function(code, state) {
                tools.debug("Keyboard: key", (state ? "pressed:" : "released:"), code);
EOF
```

mdevaev, 2024-12-30:
I made the official solution, it's already available. Happy New Year, dear users.

[image]

closed 2024-12-30 by plia7