pikvm/pikvm #1338: Split audio into L and R on headset

shafiq-mohammed, 2024-06-05:
**Is your feature request related to a problem? Please describe.**
I'm usually listening to multiple audio streams from the pikvm, so it would be really helpful if I can configure one to output volume on my left ear and the other to output volume on my right ear.

**Describe the solution you'd like**
See above! So basically two volume sliders, one for the left ear, one for the right ear. 

**Describe alternatives you've considered**
I'm using a physical mixer right now, but that adds to my setup and I have to switch audio out from pikvm to the mixer ever time I need to get on/off the mixer. I would rather not have a mixer.

mdevaev, 2025-02-10:
Sorry for the long reply. I'm not sure this is possible for video, but maybe I'm wrong. All I found is a way to control a balance: https://stackoverflow.com/questions/5123844/change-left-right-balance-on-playing-audio-in-javascript

mdevaev, 2025-05-13:
I've reviewed this issue and I think it's somewhat out of scope of the problems we're solving. But I can recommend you an extension for Chrome that does exactly what you need. If you use Firefox, there's probably an extension with similar features there too.

https://chromewebstore.google.com/detail/sound-adjustment/nebbehiamgpclkolmehcolhnfkkagdmp

closed 2025-05-13 by mdevaev