pikvm/pikvm #1343: Jump Desktop does not receive all key combos (Control+Shift+ ___ )

DonaldAHays, 2024-06-13:
**Describe the bug**
When using Jump Desktop to VNC in, some key combos are not being sent. 

**To Reproduce**
While using Jump Desktop, VNC into PiKVM. 
Control+Shift+P will not transfer, but Command+P does. 
Control+Shift+P *does* transfer with TigerVNC. 

Using Jump's "Key Conversion" does not fix anything. The conversion is also not transferred. 

**Expected behavior**
Key combos are captured. 

**Desktop (please complete the following information):**
 - OS: Mac 
 - Version: 14.5 (23F79)
 - VNC client Jump Version 8.10.8 (81008)

**PiKVM info:**
  - KVMD version: 
            kvmd 3.333-1
            kvmd-fan 0.30-1
            kvmd-oled 0.27-1
            kvmd-platform-v2-hdmi-rpi4 3.333-1
            kvmd-webterm 0.50-1
  - uStreamer version: ustreamer 6.11-1
  - Linux kernel: `6.6.21-4-rpi #1 SMP Wed Mar 13 20:36:42 UTC 2024 armv7l GNU/Linux`

**Additional context**
TigerVNC is not an option because it does not capture input from Talon Voice (voice dictation software for those with typing difficulties).

rudironsonijr, 2024-07-16:
I'm facing issues with typing in Jump too. Accents don't work on MacOS client (I'm connected through Jump on a MacOS host).
Tried pretty much everything on PiKVM and Jump's config without success.

plia7, 2024-10-28:
Could it be an issue with the Mac OS Jump Desktop client only? Because when I try to send Ctrl+Alt+P through my iOS Jump Desktop app it seems to work fine. Do you have it to try?

mdevaev, 2024-11-29:
PiKVM doesn't do anything with the keys it receives from the client. Since other clients (and even Jump on other platforms) are working, this is a problem on the Jump side. Jump changes mapping when connected to different OS. You need to set up the correct mapping or use the correct keyboard shortcuts for the existing mapping.

[image]
[image]

I don't have a Jump license to give more detailed advice, but I point out again that this is a problem in Jump.

closed 2024-11-29 by mdevaev