pikvm/pikvm #1346: Add jumper for reset CMOS

akhamar, 2024-06-15:
**Is your feature request related to a problem? Please describe.**
Doing remote configuration without been able to clear CMOS is a PITA. Sometimes server/PC hang after a reboot on bios config. Been able to clear CMOS remotely would be a huge thing !

**Describe the solution you'd like**
You just need to use two more connector on the PI, exactly like power switch / reset switch, but for a clear CMOS option.

**Describe alternatives you've considered**
N/A, expect maybe using two pin on the PIN to control that but uneasy.

**Additional context**
N/A

akhamar, 2024-06-15:
Related to https://github.com/pikvm/pikvm/issues/1252 but as a feature with hardware connector and UI menu under the proper category.

mdevaev, 2024-06-16:
Hello. All you need is using a hardware relay with [GPIO functions with menu](https://docs.pikvm.org/gpio). This is a rare enough case to be necessary in the UI as a dedicated element. GPIO exists precisely so that you can make some kind of custom for your usecases.

closed 2024-06-16 by mdevaev