pikvm/pikvm #1348: Docs: Multiport KVM over IP Table ezcoo: Typo & missing Link for KVMU3P

patrickschneider, 2024-06-17:
The Table at https://docs.pikvm.org/multiport/ should contain the Link to the ezcoo Page ( https://docs.pikvm.org/ezcoo/ ) for every Product that needs the special handling. I bought the KVMU3P KVM Switch because (according to the Table) it does not need special Handling, but could not get it to work and wasted hours on troubleshooting it.
It was only later after I accidently stumbled upon the Main ezcoo Page ( https://docs.pikvm.org/ezcoo/ ) where the KVMU3P was mentioned to need special handling.
Maybe this helps other from making the same mistake and save them some time ;-)

Also (minor type: eccoo instead of ezcoo) "eccoo EZ-SW41HA-KVMU3P 4x1 switch"

arch1mede, 2024-06-20:
Hi, thank you for the feedback, here is my feedback for that page:

The first reference to the ezcoo, since it does require setup/handling, has a link to the ezcoo page
The other 2 links for the MU3P versions, do not require any setup, we have gotten feedback from many other users with that same exact model who have not needed any other setup steps so there is nothing to link.

Can you point out where in the docs that the MU3P needs additional setup/handling? We would like to correct it.

patrickschneider, 2024-06-21:
Hello,

the 4 Port MU3P Version (as well as the MU3L Version) does need the additional USB-A to USB-Micro Cable from a Pi-USB-Port to the "Firmware/Control" Micro-USB Port of the KVM-Switch. 
The Page https://docs.pikvm.org/ezcoo/ also states this:
"The instructions here were tested with the [ezCoo SW41HA HDMI 4x1 switch](https://www.easycoolav.com/products/hdmi20-switch-4x1-with-usb20-kvm-4-port-usbsupport-4k60hz-444-and-hdr-audio-breakout) [ezCoo EZ-SW41HA-KVMU3L 4x1 switch](https://www.easycoolav.com/products/hdmi20-switch-4x1-with-usb30-kvm-3-port-usbsupport-4k60hz-444-and-hdr-audio-breakout-36) OR [ezCoo EZ-SW41HA-KVMU3P 4x1 switch](https://www.amazon.com/gp/product/B09ZKZK7ZB)."
[image]

But the 4 Port MU3P Entry on the Table on https://docs.pikvm.org/multiport/ does not contain a Link to the Page https://docs.pikvm.org/ezcoo/  
[image]
The 8 Port MU3P indeed does not seem to need special setup, so that's fine.

-> Adding the Link to the https://docs.pikvm.org/ezcoo/ page for the 4 Port MU3P Entry "Notes" column would be great
-> The 4 Port MU3P Entry Name in the Table also contains a small typo (eccoo instead of ezcoo)
Thank you!

mdevaev, 2024-07-24:
Fixed.

closed 2024-07-24 by mdevaev