pikvm/pikvm #1352: Paste fails sometimes: "HID paste error:"

yurivict, 2024-06-18:
**Describe the bug**
[image]

The text is very regular, just ASCII characters.

mdevaev, 2024-06-18:
Please check the browser console & network connections.

arch1mede, 2024-06-20:
Hi, can you provide details so we can try and reproduce this? I am unable to repo this issue. No need to provide anything sensitive, you can make this similar.

yurivict, 2024-06-20:
I can't reproduce this on purpose now.
But this happened to me twice in the last month.
I will look at the browser console next time when I see it.

mdevaev, 2024-06-21:
Okay.

yurivict, 2024-07-31:
The browser console gets an error like these every time:
[image]

yurivict, 2024-07-31:
The Network connections window gets one message like these every time:
[image]

mdevaev, 2024-08-01:
I don't understand why an SSL error occurs if the rest of the interface is working. Do you have any ideas?

mdevaev, 2024-10-12:
Do you still have this problem after all updates?

yurivict, 2024-10-13:
I will update, test, and will reopen this issue if it would come up again.

closed 2024-10-13 by yurivict

mdevaev, 2024-10-13:
Okay.