pikvm/pikvm #1355: Asrock BIOS is zoomed in on V3

chriexpe, 2024-06-22:
**Describe the bug**
This only happens when I access the BIOS, otherwise the video size is just fine.
Also you can note that the mouse  cursor isn't over mine, always quite a bit to the side.

**GIF**
[image]

**Desktop (please complete the following information):**
 - MOBO: Z690 Phantom Gaming 4/D5
 - BIOS: 16.03 (latest)
 - Firefox
 - Version 127.0.1

**PiKVM info:**
  - Raspberry Pi 4b 8GB
  - PiKVM platform: Geekworm-open source kvm-a3
  - KVMD version: kvmd 3.284-1 | kvmd-fan 0.26-1 | kvmd-oled 0.26-1 | kvmd-platform-v3-hdmi-rpi4 3.284-1 | kvmd-webterm 0.44-1
  - uStreamer version: 5.45-1
  - Linux kernel: Linux pikvm 6.1.61-1-rpi-ARCH #1 SMP Fri Nov  3 20:48:52 MSK 2023 armv7l GNU/Linux

**Additional context**
Couldn't change EDID `error: unrecognized arguments: --import-preset=v3.1080p-by-default`

mdevaev, 2024-06-22:
> PiKVM platform: Geekworm-open source kvm-a3

This is not PiKVM V3. We don't support third-party hardware. Please contact the manufacturer.

closed 2024-06-22 by mdevaev

sqanch, 2025-02-19:
I have the exact same issue with the official PiKVM v3 hat. The described behavior occurs only in the BIOS, once the OS is booted everything works fine. Any suggestions on how to fix it?

mdevaev, 2025-02-19:
Hello. Please try this: https://docs.pikvm.org/edid/#force-1080p-by-default-on-pikvm-v0

chriexpe, 2025-02-19:
> Hello. Please try this: https://docs.pikvm.org/edid/#force-1080p-by-default-on-pikvm-v0

Yup, now it's working perfectly, thank you! @sqanch thank you too for bringing this up

[image]

sqanch, 2025-02-23:
Works for me as well, thank you very much!