pikvm/pikvm #1356: Intermittent high packets loss (85% jitter) through Tailscale in specific PiKVM node which results in a black screen

plia7, 2024-06-25:
**Describe the bug**
Following the fix for intermittent black screen in WebRTC in external network made in: https://github.com/pikvm/pikvm/issues/1084 made by @RSATom.
The "invalid peer connection" error is not showing anymore, but I still get the intermittent black screen, although this time it appears to happen because of high packets loss.

**To Reproduce**

1. Connect to pikvm from an external network using tailscale installed on the external network connecting machine.

2. Stay connected and use it for few hours.

**Expected behavior**
Being able to stay connected and use it for few hours without any interruptions assuming stable and fast internet connection on both ends. 

I pointed the speedtest in client to the host local speedtest server and got:

[image]

So connectivity is good between client and host. The general speed test in host pc is:

[image]

**Actual behavior**
Intermittent black screen in WebRTC starts appearing after some time (could be after minutes or hours) where it works fine until that point, then prior to the black screen it might appear like the mouse movement is slower to react (although no observed internet quality reduction on both ends). It can stay like that for couple of hours or more until the problem is resolved by itself.

[image]

These are my runtime settings (I tried to change gop to 0 but that didn't help):
[image]

**Additional context**
I have two pikvms connected through tailscale to the same host machine, and the issue could happen with pikvm1, but pikvm2 would work fine (just as described originally in https://github.com/pikvm/pikvm/issues/1084.).
When the issue happens, I ran the iperf3 server through pikvm 1 ssh (iperf3 --server) and then ran the iperf3 client on the client machine (iperf3 --udp --client pikvm --bandwidth 5M -R) and this was the result:

```
[ ID] Interval           Transfer     Bandwidth       Jitter    Lost/Total Datagrams
[  4]   0.00-1.02   sec  88.0 KBytes   710 Kbits/sec  56149221283.709 ms  32/43 (74%)
[  4]   1.02-2.00   sec  88.0 KBytes   729 Kbits/sec  27607544377.343 ms  78/89 (88%)
[  4]   2.00-3.00   sec  88.0 KBytes   724 Kbits/sec  13574124279.860 ms  70/81 (86%)
[  4]   3.00-4.01   sec  80.0 KBytes   652 Kbits/sec  7119091711.364 ms  73/83 (88%)
[  4]   4.01-5.01   sec   120 KBytes   980 Kbits/sec  2703919410.526 ms  88/103 (85%)
[  4]   5.01-6.00   sec  88.0 KBytes   727 Kbits/sec  1329467736.519 ms  53/64 (83%)
[  4]   6.00-7.01   sec  96.0 KBytes   782 Kbits/sec  612820321.101 ms  85/97 (88%)
[  4]   7.01-8.01   sec   136 KBytes  1.11 Mbits/sec  204571412.982 ms  52/69 (75%)
[  4]   8.01-9.01   sec  88.0 KBytes   723 Kbits/sec  100584089.821 ms  74/85 (87%)
[  4]   9.01-10.01  sec  72.0 KBytes   588 Kbits/sec  56269244.973 ms  46/55 (84%)
- - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bandwidth       Jitter    Lost/Total Datagrams
[  4]   0.00-10.01  sec  6.74 MBytes  5.65 Mbits/sec  27666619.152 ms  709/838 (85%)
[  4] Sent 838 datagrams

iperf Done.
```

So yes, it's a big jitter loss at 85%. And I can't basically to restore the jitter to go down even if I try to disconnect from tailscale and reconnect again.. Interestingly pikvm2 doesn't have a high jitter. So somehow the pikvm tailscale connection becomes unstable at some periods, but it happens with some specific tailscale pikvm node but not the other pikvm node(s) although it's the same distance/destination machine. 

- So how do I reset the tailscale pikvm connection of pikvm 1 so it goes to same level as pikvm 2?

- When this issue happens, the screen goes black for long periods of times and I can't really use it, but the second screen pikvm seems to be ok... which doesn't make sense.

- Basically the issue I believe is: tailscale associated with a particular pikvm packets loss becomes high resulting in low kbps (less than 1000kbps) which causes black screen for hd resolution - but it doesn't mean all nodes in tailscale become this high in packets loss like second pikvm, even with same distance.... does it sound like a tailscale issue or pikvm issue?

- So somehow through tailscale in one pikvm node it develops 85% jitter and persists for hours, while the second pikvm node is totally fine and stable (picture is showing and mouse/keyboard is responsive):

[image]

- Given that Tailscale is the recommended solution to use in conjunction with pikvm, is there any way to get them work together without these interruptions or are other alternative mesh-vpn solutions that won't have the same issue? Because it makes it unusable when this issue happens.
- Could pikvm somehow cause the tailscale jitter go very high (bug in the code)?

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
Windows 11
 - Browser [e.g. chrome, safari]
Google Chrome Version 126.0.6478.115 (Official Build) (64-bit) (cohort: Stable) 
 - Version [e.g. 22]
 - VNC client (if used)

**PiKVM info:**
  - Raspberry Pi board version [e.g. RPi 4]
PiKVM v3 Pre-Assembled
  - PiKVM platform [e.g. v2-hdmi]
  - Video capture type [e.g. CSI bridge]
  - KVMD version: `pacman -Q | grep kvmd`
kvmd 3.321-1
kvmd-fan 0.30-1
kvmd-oled 0.26-1
kvmd-platform-v3-hdmi-rpi4 3.321-1
kvmd-webterm 0.48-1
  - uStreamer version: `pacman -Q | grep ustreamer`
ustreamer 6.4-1
  - Linux kernel: `uname -a`
Linux pikvm 6.6.21-4-rpi # 1 SMP Wed Mar 13 20:36:42 UTC 2024 armv7l GNU/Linux

Thanks.

RSATom, 2024-06-26:
2.25 mbps is not good speed, but it depends what direction it is... Also, I would recommend you check how good is your physical connection of pikvm. For example It's possible you have broken Ethernet cable somewhere.

plia7, 2024-06-26:
> 2.25 mbps is not good speed, but it depends what direction it is... Also, I would recommend you check how good is your physical network connection of pikvm. For example It's possible you have broken Ethernet cable somewhere. 

Where do you see 2.25mbps?
The pikvm has an ethernet cable connected to it. I never had this issue when connecting through the local network.

Thanks.

RSATom, 2024-06-26:
[image]

plia7, 2024-06-26:
> [image]
> 
> 

As discussed, this is the uploading to the host pc side from connecting remote pc which you said is ok.

mdevaev, 2024-06-26:
I have not encountered such software problems. Perhaps this is what we have told: the problem of the CM4 connector.

plia7, 2024-06-26:
> CM4 connector

I didn't install the cm4 module since it's the pikvm v3 pre-assembled version - How do I even reinstall it?
Thanks.

plia7, 2024-06-26:
> I have not encountered such software problems. Perhaps this is what we have told: the problem of the CM4 connector.

I just reproduced it with pikvm2, with 87% jitter, so I doubt it's a physical issue with pikvm itself.

mdevaev, 2024-06-28:
So probbly it's just a local network or tailscale conditions.

plia7, 2024-06-28:
> So probbly it's just a local network or tailscale conditions.

just experienced the problem in pikvm 1, sent it to a reboot, and the problem seem to be gone. So is it most likely pointing to tailscale conditions?

mdevaev, 2024-06-28:
Yes, maybe.

plia7, 2024-06-28:
> Yes, maybe.

Is there a way to reboot tailscale service without sending pikvm for a reboot? To rule out potential pikvm issue?

mdevaev, 2024-06-29:
* `systemctl restart tailscaled`
* `ip link set eth0 down; sleep 20; ip link set eth0 up`

mdevaev, 2024-07-01:
@RSATom told me that you found out that you had a problem with the Internet provider.

closed 2024-07-01 by mdevaev

plia7, 2024-07-01:
> @RSATom told me that you found out that you had a problem with the Internet provider.

Well yes it appears this way but somehow I'm still able to run both pikvms despite not having supposedly enough bandwidth. The issue just happened, I ran "" and it appears to correct it. But I'll keep monitoring and analyzing.

Thanks.