pikvm/pikvm #1357: Adding monitor/audio vendors/SN/model/product id to override.yaml

mareksagan, 2024-06-27:
**Is your feature request related to a problem? Please describe.**
Using the kvmd-edidconf is not very comfortable if you need to manage multiple devices. It would be much easier to share the same override.yaml file and have the ability to set the settings there. I don't need a specific EDID setting, vendor/serial number/model/product id should be displayed in the Device Manager and settings.
[image]
[image]

**Describe the solution you'd like**
Please add this customization from kvmd-edidconf to override.yaml:
[image]
In a manner similar to this one:
[image]

**Describe alternatives you've considered**
None, it's simple enough, I think.

**Additional context**
Thank you for your help.

mdevaev, 2024-06-27:
Hello. Why you can't just copy /etc/kvmd/tc358743-edid.hex after all changes made by kvmd-edidconf?

mareksagan, 2024-06-28:
Because then I have to copy two files and it's less convenient. As I mentioned, you're probably right that it's not a problem if you have one device. I have many.

mdevaev, 2024-06-28:
I'll think about whether it's possible to do it the way you want. The fact is that EDID is a fairly low-level thing that is used in V4L2 even before KVMD starts working.

mareksagan, 2024-06-28:
Ok, I understand. Maybe in the meantime you could guide me through the standard EDIDs please? I mean, what resolution, refresh frequency and monitor make/model do they set when I use them? I couldn't find the exact description for them.
[image]

mdevaev, 2024-06-28:
V4 uses default 1080p resolution and supports 60Hz, also supports 1920x1200. V0-V3 uses default 720p, but supports 1080p at 50Hz.

mdevaev, 2024-08-30:
Given the other features we have planned, this will not be implemented because we need flexible configuration of EDID data. I would recommend rolling out configs using tar archives.

closed 2024-08-30 by mdevaev