pikvm/pikvm #1358: kvmd-edidconf --set-monitor-serial=[MONITOR_SERIAL] doesn't work on PiKVMv3

mareksagan, 2024-06-28:
**Describe the bug**
I cannot change the Monitor Serial number using kvmd-edidconf tool.

**To Reproduce**
Steps to reproduce the behavior, like:
1. rw
2. kvmd-edidconf --set-monitor-serial=123A
3. This error occurs:
Traceback (most recent call last):
  File "/usr/bin/kvmd-edidconf", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/usr/lib/python3.12/site-packages/kvmd/apps/edidconf/__init__.py", line 377, in main
    getattr(edid, cmd)(value)
  File "/usr/lib/python3.12/site-packages/kvmd/apps/edidconf/__init__.py", line 188, in set_monitor_serial
    self.__set_dtd_text(0xFF, "Monitor Serial", text)
  File "/usr/lib/python3.12/site-packages/kvmd/apps/edidconf/__init__.py", line 195, in __set_dtd_text
    index = self.__find_dtd_text(d_type, name)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/site-packages/kvmd/apps/edidconf/__init__.py", line 204, in __find_dtd_text
    raise NoBlockError(f"Can't find DTD {name}")
kvmd.apps.edidconf.NoBlockError: Can't find DTD Monitor Serial

**Expected behavior**
Monitor serial should change but it doesn't.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Win11
 - ssh (Ubuntu Subsystem)

**PiKVM info:**
  - Raspberry Pi board version: RPi4
  - PiKVM platform: v3
  - Video capture type: CSI bridge
  - KVMD version: kvmd 4.2-1,kvmd-fan 0.31-1,kvmd-oled 0.27-1,kvmd-platform-v3-hdmi-rpi4 4.2-1,kvmd-webterm 0.50-1
  - uStreamer version: ustreamer 6.12-1
  - Linux kernel: Linux pikvm 6.6.21-4-rpi #1 SMP Wed Mar 13 20:36:42 UTC 2024 armv7l GNU/Linux

**Additional context**
Thank you for your help.

mdevaev, 2024-06-28:
`Serial` and `Monitor serial` are two different type of blocks. Use `--set-serial`, like: `kvmd-edidconf --set-monitor-name=TOSHIBA --set-mfc-id=TTP --set-product-id=34953 --set-serial=2290649089 --apply`

closed 2024-06-28 by mdevaev

mareksagan, 2024-06-28:
But I want to be able to set a monitor serial. How do I do that?

mdevaev, 2024-06-28:
kvmd-edidconf is able to configure only those blocks that are already in EDID, with the exception of Audio. This is not a full-fledged editor. If you need more, use the AW EDID Editor.

mareksagan, 2024-06-28:
I tried opening the tc358743-edid.hex file with AW EDID Editor but it gives me this error... Do you know how to fix this one please?
[image]

mdevaev, 2024-06-28:
Please follow the guide: https://docs.pikvm.org/edid