pikvm/pikvm #1361: Browser tab name customization for the PiKVM session

mareksagan, 2024-07-01:
**Is your feature request related to a problem? Please describe.**
I want to be able to change the name that appears on the tab of the PiKVM session.
[image]

**Describe the solution you'd like**
Just add a override.yaml field that corresponds to the tab session name:
kvmd:
    tab_name: 'Happy Session'

It would change the name 'PiKVM Session: localhost:localdomain' to "Happy Session'

**Describe alternatives you've considered**
None.

**Additional context**
Thank you.

mdevaev, 2024-07-07:
Hello. You can change `localhost:localdomain` part in /etc/kvmd/meta.yaml

mdevaev, 2024-07-17:
Solved, I guess.

closed 2024-07-17 by mdevaev