pikvm/pikvm #1365: Audio not working on PiKVM-A3

BluishHumility, 2024-07-04:
Hello,

I am having some difficulty getting audio working. I have a PiKVM-A3 with a Raspberry Pi 4B. I am using the H.264/WebRTC video mode in a web browser. Everything is working well, except I can't seem to figure out how to get audio working.

I have gone through this document here: https://docs.pikvm.org/audio/

I have not removed the audio jumpers, nor deleted the `dtoverlay=tc358743-audio` line in `/boot/config.txt`. Here is the entirety of `/boot/config.txt` if it is helpful:
```
# See /boot/overlays/README for all available options
initramfs initramfs-linux.img followkernel

hdmi_force_hotplug=1
gpu_mem=128
enable_uart=1
dtoverlay=tc358743
dtoverlay=disable-bt
dtoverlay=dwc2,dr_mode=peripheral
dtparam=act_led_gpio=13

# HDMI audio capture
dtoverlay=tc358743-audio

# SPI (AUM)
dtoverlay=spi0-1cs

# I2C (display)
dtparam=i2c_arm=on

# Clock
#dtoverlay=i2c-rtc,pcf8563,wakeup-source
dtoverlay=i2c-rtc,ds1307

# Passthrough
dtoverlay=vc4-kms-v3d
disable_overscan=1
```
The `audio` block mentioned in the documentation is present in `/etc/kvmd/janus/janus.plugin.ustreamer.jcfg`:
```
video: {
    sink = "kvmd::ustreamer::h264"
}
audio: {
    device = "hw:0,0"
    tc358743 = "/dev/kvmd-video"
}
```
Audio support is enabled in the EDID hex:
```
[root@pikvm ~]# kvmd-edidconf --set-audio=yes
Manufacturer ID: LNX
Product ID:      0x7771 (30577)
Serial number:   [serial number]
Monitor name:    PiKVM V3
Monitor serial:  [monitor serial]
Audio:           yes
```
I have rebooted after setting this, then increased the "Audio volume" in the web UI to max, but I do not get any audio.

* Audio works normally on the laptop the PiKVM is connected to (out of the ordinary speakers);
* The laptop recognizes the connected HDMI device as a valid audio output, and visually indicates that audio is playing on that device;
* I have tried multiple HDMI cables just in case;
* I tested two different computers (to bring up the web page), both of which have audio working normally and can get audio in the browser on other sites;
* Chromium and Firefox were both tested, on both Arch Linux with i3 and EndeavourOS with Sway.

Probably this is the smoking gun here in the journal, although I have not been able to figure out what the problem is or how to resolve it:
```
[...]
kvmd.apps.janus.runner            INFO --- => JANUS WebSockets transport plugin initialized!
kvmd.apps.janus.runner            INFO --- => WebSockets thread started
kvmd.apps.janus.runner            INFO --- => Creating new session: 4652817366914926; 0xb0e00bf0
[... 94 lines cut ...]
kvmd.apps.janus.runner            INFO --- => == ustreamer/audio     -- Pipeline closed
kvmd.apps.janus.runner            INFO --- => == ustreamer/audio     -- Detected host audio
kvmd.apps.janus.runner            INFO --- => == ustreamer/audio     -- Pipeline configured on 48000Hz; capturing ...
kvmd.apps.janus.runner            INFO --- => [ERR] [src/audio.c:_pcm_thread:224] == ustreamer/audio     -- Fatal: Can't capture PCM frames: Input/output error
[...]
```
That block at the end repeats over and over and over until the end of the journal.

I did a web search for this error message and found this thread: https://github.com/pikvm/pikvm/issues/1193 It's not quite the same, but somewhat similar...however, that person isn't using PiKVM hardware.

Thanks for taking a look, please let me know if I have made a mistake somewhere or if you need any additional information.

**Desktop:**
 - OS: Arch Linux, i3wm
 - Another OS: EndeavourOS, Sway
 - Browser: Chromium 126.0.6478.114
 - Another browser: Firefox 127.0.2-2

**PiKVM info:**
  - Raspberry Pi 4 Model B Rev 1.2
  - PiKVM-A3 (SKU 100328-A3)
  - Video capture type: X630 HDMI to CSI-2 board 
  - KVMD version: 4.2-1
  - uStreamer version: 6.12-1
  - Linux kernel: 6.6.21-4-rpi

mdevaev, 2024-07-04:
Hello. You bought a clone. We guarantee the operation of our software only for proven [DIY builds](https://docs.pikvm.org/v2/) and for [our own hardware V3 and V4](https://pikvm.org/buy/). Audio is an exclusive feature of our own hardware.

Unfortunately, we do not provide any technical support to third-party/clone devices.

closed 2024-07-04 by mdevaev

BluishHumility, 2024-07-04:
This is surprising news because I thought I bought the right thing. However, a quick look at the link you posted confirms I did not. The unit I have looks similar, but not the same.

Too bad, I will certainly be more careful next time. Thanks anyway for taking a look. :slightly_smiling_face:

mdevaev, 2024-07-05:
:ok_hand:

PureIncompetence, 2025-12-31:
I've got the same problem as described in this thread, except that I bought the official `PiKVM Shield v3.3` - at least according to the print on the board and the information in the seller's shop - from one of the resellers you linked to on [your website](https://pikvm.org/buy/#pikvm-v3-hat). I've attempted to fix it in the same ways the thread opener tried to. As a browser I'm using the latest version of ungoogled chromium.

In the output of `journalctl -u kvmd-janus` I see the following lines repeating whenever the remote controlled device is playing sound:
```
[src/acap.c:_pcm_thread:172] == ustreamer/acap      -- Fatal: Can't capture PCM frames: Input/output error
ustreamer/acap      -- Capture closed
ustreamer/acap      -- Detected host audio
ustreamer/acap      -- Capture configured on 48000Hz; capturing ...
```
While audio output did work once (before I tried enabling the microphone in the override.yaml), I didnt get the mic working at all.

PureIncompetence, 2025-12-31:
Thanks for the hint (and the prompt reaction). But that toggle didnt even show up in the first place.

After like 10 reboots and trying the same thing over and over again I got it working - kind of. The audio playback works ok, with slight to modest stutters every now and then even though the device is connected via gbit ethernet. The mic however seems to be constantly stuttering. I tested it by recording the mics audio with audacity on the target device. But thats probably a seperate problem which needs to be discussed in its own issue. Besides I still have to try whether the current state (audio working more or less) will survive a few reboots.

I wish you a happy new year!