pikvm/pikvm #1368: Extending OLED life: Dim/off when user logs in

Zitt, 2024-07-11:
**Is your feature request related to a problem? Please describe.**
Been using V3 OLED non-stop (read 24/7/365) for over a year and there is a noticeable inability to read the OLED screen at 25%. I know you made some changes in #994 to help; but, I feel it's not enough or optimal. 

**Describe the solution you'd like**
I'd like some settings in /etc/kvmd/override.yaml.
Tie those settings into wither a user is logged into the kvm.

kvmd:
  oled:
    loginBrightness: 12 
    logoutBrightness: 50

when a user logs into the kvm; brightess would be set to loginBrightness. When they log out: logoutBrightness
In my case; I'd probably set it to 0 when logged in... and maybe 25 when logged out.

mdevaev, 2024-07-12:
Hi. How often do you use PiKVM compared to its inactivity?

Zitt, 2024-07-16:
> Hi. How often do you use PiKVM compared to its inactivity?

Usually daily M-F. I usually leave the "host" machine logged into the PiKvm and then remote desktop into that machine as I run my tests/interact with the machine.

mdevaev, 2025-09-22:
Yes, we have some api to do this now, see in the next update.