pikvm/pikvm #1370: Chrome doesn't offer to save password for the login form served by the PiKVM instance

yurivict, 2024-07-18:
**Describe the bug**

Chromium doesn't store the login/password combination for the local PiKVM instance.

I am not sure whether this is a bug in PiKVM or not, but I am still reporting this just in case.

Here is the log from chrome://password-manager-internals/: [log](https://freebsd.org/~yuri/password-manager-internals-2024-07-18_19-53-27.html).

It says "SSL errors present: true". This this possibly because the certificate is self-signed, but I am not sure.

**Desktop (please complete the following information):**
 - OS: FreeBSD 14.1
 - Browser chrome

**PiKVM info:**
  - Raspberry Pi board version 4

mdevaev, 2024-07-23:
Hello. It can't be fixed as I understand: https://superuser.com/questions/1241132/how-to-force-chrome-to-save-passwords-on-self-signed-or-broken-ssl

yurivict, 2024-07-23:
I see.

Thank you for this information.
I'll close this issue then.

closed 2024-07-23 by yurivict