pikvm/pikvm #1371: RPi 2 rev 1.2 H.264 encoding support

eiqnepm, 2024-07-20:
**Is your feature request related to a problem? Please describe.**
Unlike the RPi 2 rev 1.1, the RPi 2 rev 1.2 has the same SoC as the RPi 3, albeit underclocked slightly, however as far as I can tell, PiKVM does not officially support H.264 encoding on the RPi 2 rev 1.2.

**Describe the solution you'd like**
Please may the PiKVM image for the RPi 2 include H.264 encoding if the board revision is 1.2?

**Describe alternatives you've considered**
Purchasing a RPi 3 or better.

**Additional context**
https://docs.pikvm.org/v1/#required-parts
https://docs.pikvm.org/flashing_os/#download-the-image

It would be awesome if I could put my RPi 2 rev 1.2 to use!

mdevaev, 2024-07-23:
I didn't test H.264 with Pi 2 and I'm not sure if this working at all :/

eiqnepm, 2024-07-23:
> I didn't test H.264 with Pi 2 and I'm not sure if this working at all :/

When I have acquired a HDMI-CSI-2 bridge, please may you be able to create an SD card image for the RPi 2 rev 1.2 with H.264 enabled for me to test?

If it works the image can be distributed for others who have a RPi 2 rev 1.2.

mdevaev, 2024-07-23:
You can try it yourself with very few changes:
1. Add this to `/etc/kvmd/override.yaml`:
```yaml
kvmd:
    streamer:
        h264_bitrate:
            default: 5000
        cmd_append:
            - "--jpeg-sink=kvmd::ustreamer::jpeg"
            - "--jpeg-sink-mode=0660"
            - "--h264-sink=kvmd::ustreamer::h264"
            - "--h264-sink-mode=0660"
            - "--h264-bitrate={h264_bitrate}"
            - "--h264-gop={h264_gop}"
vnc:
    memsink:
        jpeg:
            sink: "kvmd::ustreamer::jpeg"
        h264:
            sink: "kvmd::ustreamer::h264"
```
2. Add line `gpu_mem=128` to `/boot/config.txt`
3. Add `cma=128` to `/boot/cmdline.txt`
4. Run `systemctl enable kvmd-janus`
5. `reboot`

eiqnepm, 2024-07-23:
> You can try it yourself with very few changes:
> 1. Add this to `/etc/kvmd/override.yaml`:
> ```yaml
> kvmd:
>     streamer:
>         h264_bitrate:
>             default: 5000
>         cmd_append:
>             - [...]

Thanks!

I will report back on the outcome when I get the HDMI-CSI bridge and test this.

mdevaev, 2024-08-29:
Sup?

eiqnepm, 2024-08-29:
> Sup?

I had issues getting a HDMI-CSI bridge so I ended up just forgetting about it. I've since pre-ordered another solution.

This isn't an issue for me anymore, but I hope the issue will still be useful to anyone that may need it in the future, feel free to close it for now though, thanks.

closed 2024-08-29 by mdevaev