pikvm/pikvm #1372: PIKVM V4-Public net work

reddeppa-eddala, 2024-07-21:
Right now PIKVM is connected with local network which is my home internet and i could connect my laptop remotely and no issue.

when i want use different network which means different internet but it is not working and checked with google and confirmed that using tail scale VPN is possible and followed those steps and updated at os Level but no luck.

Please refer this link -https://tailscale.com/kb/1292/pikvm.

Now my question to you : Is it possible use PIKVM device with different network? if yes please share the steps and also i am ready for online support .

mdevaev, 2024-07-21:
For support questions please use our discord: https://discord.gg/bpmXfz5

reddeppa-eddala, 2024-07-22:
I have registered in this link -[ttps://discord.gg/bpmXfz5](https://discord.gg/bpmXfz5) and not sure how to access it . 

Could you please reply to my query 

Right now PIKVM is connected with local network which is my home internet and i could connect my laptop remotely and no issue.

when i want use different network which means different internet but it is not working and checked with google and looks like  using tail scale is possible and followed those steps as per the below link  and updated at os Level but no luck.

Please refer this link -https://tailscale.com/kb/1292/pikvm.

Now my question to you : Is it possible use PIKVM device with different network? if yes please share the steps and also i am ready for online support .

arch1mede, 2024-07-22:
Hi, yes I can confirm that with a default image, this should work on any network that uses DHCP that you place this on, we would need to see more details as to why this isn't working which Discord would help us with.

reddeppa-eddala, 2024-07-22:
Is it possible to give online support from your end and will play for that

reddeppa-eddala, 2024-07-22:
HI, 

Any PIKVM device would suggest  where public net work should support without tail scale 

Regards, 
Reddeppa

mdevaev, 2024-07-23:
Online support is provided in the chat, I gave you the link earlier (https://discord.gg/bpmXfz5). PLease use the web browser and ask a question there.

closed 2024-07-23 by mdevaev

reddeppa-eddala, 2024-07-24:
Thank you so much for your support. Now i am able to login with public network using tail scale. however  the tailscale is required to install in the laptop but my office laptop is not allowing to install the tail scale to due to some restrictions.

Instead of Tail scale, is it possible to use TeamViewer . Please confirm. if yes please share the steps 

Regards, 
Reddeppa

mdevaev, 2024-07-24:
You can't use TeamViewer for this.

reddeppa-eddala, 2024-07-24:
Other than Tailscale, Any other software will support ?

Whenever i want to connect with public IP, Is tail scale must be installed in the laptop?

mdevaev, 2024-07-24:
Please, use the chat for support questions. Github is only for development and bug reports.

reddeppa-eddala, 2024-07-24:
Please send the link for chat

mdevaev, 2024-07-24:
I gave you the link twice, see above.