pikvm/pikvm #1373: Allow for mouse translation based on local screen size

jordan30001, 2024-07-22:
**Is your feature request related to a problem? Please describe.**
When connecting to a device that has a mirrored output which is larger than than what the PiKVM supports (e.g. the monitor connected to the KVM is outputting 4k, but the GPU on the device is downscaling the output to the KVM to 1080p) the bounding box for the mouse only works within the scaled area (1080p) which causes weird mouse movement when outside of the bounding box and jagged mouse movements when in Absolute Mouse Mode.
(See below image, the red area is the bounding box for the downscaled 4k output, and the black area is the rest of the screen, which is also 4k)
[image]

**Describe the solution you'd like**
Add an option to the KVMs web interface to allow for changing of the mouses resolution at runtime. (see below screenshot of where this could be added)
[image]

**Describe alternatives you've considered**
Allow for an override within the override.yml settings file to set the absolute mouse mode to use a custom resolution (e.g. 3840x2160), rather than the input size of the video input.
This solution works for me, however, for others it may not be dynamic enough comparatively to an option directly in the KVMs web interface.

arch1mede, 2024-07-27:
I'd wait for the dev to make a comment but please be sure you understand that PiKVM cannot influence the target system, this is the same as if you put a 1080p monitor on the system and asked the monitor manufacturer the same question. This is nothing more than an IP reachable monitor/kb/mouse.

jordan30001, 2024-07-28:
Based on the current description of how the KVM works, this should work.

["In absolute mode, the input device transmits the exact coordinates (X,Y) where the cursor should be moved. This is how touchscreens or drawing tablets work." ](https://docs.pikvm.org/mouse/#:~:text=In%20absolute%20mode%2C%20the%20input%20device%20transmits%20the%20exact%20coordinates%20(X%2CY)%20where%20the%20cursor%20should%20be%20moved.%20This%20is%20how%20touchscreens%20or%20drawing%20tablets%20work.)
If this is the case then no modification to the target system is required, other than setting the remote systems resolution to 4k to the monitor and scaling the resolution to 1080p to the kvm (which i am not expecting the KVM to do)

If we say the Viewport is 1920x1080 and the target system is 3840x2160:
We can calculate the scales with the following pseudocode
```
customResolution = (Width: 3840, Height: 2160) // settings file on kvm, or settings in the web interface
Viewport = (Width: 1920, Height: 1080) //the web interfaces viewport resolution
Mouse = (X: 100, y: 150) // current mouse position in the web interfaces viewport
xScale = customResolution.Width / ViewPort.Width // 2
yScale = customResolution.Height / ViewPort.Height // 2
translatedX = Mouse .X * xScale // 200
translatedY = Mouse.Y * yScale // 300

sendAbsolutePosition(translatedX, translatedY)
```
If in the above pseudocode the viewport is also at a resolution of 3840x2160 then the scale would be a 1:1 ratio and would remove the jagged movement of the mouse on the target system.