pikvm/pikvm #1374: PIKVM support requested. In addition can't log to discord

frenchyinmunich, 2024-07-25:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

frenchyinmunich, 2024-07-25:
Hello,
I have bought a PiKVM kit V4. It is mounted and now, I would like to have some support to correctly flash etc...
I feel lost. In addition it is impossible to log on the Discord chat.
Could you please help me?
Thanks,
Francois

mdevaev, 2024-07-25:
V4 is a ready-made device. What kit did you buy? Please take a photo.

frenchyinmunich, 2024-07-25:
Could I call you please?Thel ist of component I bought is attached

mdevaev, 2024-07-25:
Github doesn't handle attaches. Please copy here.

frenchyinmunich, 2024-07-25:
> Github doesn't handle attaches. Please copy and paste here.

frenchyinmunich, 2024-07-25:
can I call U?

mdevaev, 2024-07-25:
We provide support only in the discord. At least in the github.

frenchyinmunich, 2024-07-25:
But your invitation is refused

frenchyinmunich, 2024-07-25:
It says the invitation can't be accepted.Why?

frenchyinmunich, 2024-07-25:
Here is the reason I can't connect:Invalid Invite Links

| 
| 
| 
|  |  |

 |

 |
| 
|  | 
Invalid Invite Links

Let me innnnnnn! You finally got an invite to a server, but you’re getting hit with an error message and can’t j...
 |

 |

 |

frenchyinmunich, 2024-07-25:
Please help me

mdevaev, 2024-07-25:
Take a photo of your kit here. Please use github interface for attaching photos, don't use email client.

mdevaev, 2024-07-25:
Please use github interface for attaching photos, don't use email client. I don't see your photos.

frenchyinmunich, 2024-07-25:
[image]

frenchyinmunich, 2024-07-25:
just sent on GitHub

frenchyinmunich, 2024-07-25:
[image]

frenchyinmunich, 2024-07-25:
[image]

frenchyinmunich, 2024-07-25:
[image]

frenchyinmunich, 2024-07-25:
[image]

frenchyinmunich, 2024-07-25:
[image]

frenchyinmunich, 2024-07-25:
[image]

frenchyinmunich, 2024-07-25:
[image]

mdevaev, 2024-07-25:
I asked our support team to help you.

frenchyinmunich, 2024-07-25:
Please, tell them to hurry up

frenchyinmunich, 2024-07-25:
Hi, can the support team contact me please?

mdevaev, 2024-07-25:
Please wait.

frenchyinmunich, 2024-07-25:
I understand, but I need the device to work.

frenchyinmunich, 2024-07-25:
for my job

frenchyinmunich, 2024-07-25:
and why I can't use discord?what a stress!

mdevaev, 2024-07-25:
Until support has contacted you, you can try to follow the guide carefully: https://docs.pikvm.org/v3/

arch1mede, 2024-07-25:
Hi, github is not used for support, please join discord for [live chat](https://discord.gg/bpmXfz5). What you bought and what you are describing is not the same so we need to clarification.

arch1mede, 2024-07-25:
All of your discord pics are broken, so I cannot tell what the issue is.

frenchyinmunich, 2024-07-25:
Please, consider what I have bought

frenchyinmunich, 2024-07-25:
- UStId-Nr.: DE287421501 | WEEE-Reg.-Nr.: DE53739998    
Bundle bestehend aus Raspberry Pi 4 Modell B (4 GB), internationalemNetzteil (EU/UK/US/AU), PiKVM HAT und PiKVM Steel Case inkl.Montagematerial, OLED Display und Lüfter2409332-010Goobay Ethernet-Kabel 1m CAT610,92 €19%1,10 €CAT 6 Patchkabel, S/FTP (PiMF), schwarz, LSZH halogenfrei, Kupfer3409151Kühlkörpersatz11,09 €19%1,30 €Schwarz eloxierte Aluminium-Kühlkörper mit selbstklebendem Wärmeleitpadfür Raspberry Pi CPU und USB/LAN-IC4GB1-058572Goobay 58572 HDMI Kabel, HDMI 2.0 LC, 1,00m11,84 €19%2,19 €Series 2.0 LC High Speed HDMI Kabel mit Ethernet, 1 m, Schwarz -HDMI-Stecker (Typ A) > HDMI-Stecker (Typ A)5GB1-055468Goobay 55468 USB-C Kabel, USB 2.0, 1,80m13,02 €19%3,59 €USB 2.0 Kabel USB-C auf USB A, schwarz, 1.8 m - geeignet für Geräte mitUSB-C Anschluss6DHL Express DE116,72 €19%19,90 €7Voucher1-6,72 €19%-8,00 €Gesamtpreis netto241,24 €zzgl. 19% MwSt.45,84 €Gesamtpreis brutto287,08 €Zahlung PayPal Express, 17.07.2024-287,08 €offener Betrag0,00 €Fortsetzung auf Seite 2
   - Marco Wünschmann Haid-und-Neu-Str. 7 76131 Karlsruhe Germany fon +49 721 909819-90 mail ***@***.*** web www.welectron.com Haid-und-Neu-Str. 7  6131 Karlsruhe

frenchyinmunich, 2024-07-25:
What a nightmare, and I still can't connect to discord....

arch1mede, 2024-07-25:
There is no other means of support for PiKVM other than Discord, I have considered what you have purchased but most would be cleared up over live chat.

frenchyinmunich, 2024-07-25:
It is still impossible to connect to discord!!!!!

arch1mede, 2024-07-25:
Please try on Chrome, incognito mode.

arch1mede, 2024-07-25:
The only other option is to contact Welectron for support, they in turn can join Discord for assistance but most of what's needed should be in the [v3 document](https://docs.pikvm.org/v3/) and [First_steps document](https://docs.pikvm.org/first_steps/).

frenchyinmunich, 2024-07-25:
I am connected to discordcould you please telle me now what should I do?

arch1mede, 2024-07-25:
Please post your question in the #support channel.

frenchyinmunich, 2024-07-25:
tell me your pseudo on discord please

arch1mede, 2024-07-25:
To start off, post your question in the #support channel

frenchyinmunich, 2024-07-25:
My account on Discord is: Polykobol

arch1mede, 2024-07-25:
Seems you clicked on the wrong discord link, please use [this one](https://discord.gg/bpmXfz5).

closed 2024-07-25 by mdevaev