pikvm/pikvm #1375: WLAN connectivity lost after upgrade today

kusmi, 2024-07-25:
**Describe the bug**
Today I updated the pikvm (pikvm DIY V2 using a , on a raspberry-pi Zero 2 W with HDMI CSI bridge) using "pikvm-update" - I try to be up-to-date, so my last update was approx. last week, so no large update I guess.
After update today the pikvm lost network (WLAN) connection and I can't connect to device anymore.

I then formatted the SD card and setup everything new again. pikvm was on network again (with a very old version) and I did upgrade again "pikvm-update" (took a while) - and the same behavior: After update the pikvm does no join the WLAN again.

If I connect the pikvm to my mac - after some time a new USB device connects (Composite KVM Device with serial CAFEBABE), so it looks like pikvm still starts - it can just not connect to WLAN anymore and I can't find it on my WLAN router, also arp scans did not find the device on my network.

**To Reproduce**
1. use raspberry-pi zero 2 w (perhaps other devices as well?)
2. upgrade system as root: "pikvm-update"
3. system reboots
4. loss of wlan access, so no SSH or web access

**Expected behavior**
after upgrade wlan connectivity should not be missing - I had no problems in the past with upgrades

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Macosx
 - Browser: Safari, Chrome
 - Version [e.g. 22]
 - VNC client (if used)

**PiKVM info:**
  - Raspberry Pi board version raspberry pi zero 2 w
  - Video capture type: CSI bridge

**Additional context**
Add any other context about the problem here.

mdevaev, 2024-07-25:
Follow my last comment from that please:

> Did you try fresh image of the official Raspberry OS with latest updates? We need to understand this is an Arch or Raspberry upstream problem.

kusmi, 2024-07-25:
It works with latest Raspberry OS

I did:
1. Installed Raspberry PI OS (64 bit) using "Raspberry Pi Imager"
2. Did "sudo apt update" and "sudo apt full-upgrade"
3. rebooted
4. shutdown and disconnect from power (to be sure)
5. rebooted
6. it did connect to WiFi

Or do I need to perform different steps?

mdevaev, 2024-07-25:
Check 32 bit OS please.

kusmi, 2024-07-25:
ok, did the same procedure with Raspberry Pi OS 32bit (Description in Pi Imager: "A port of Debian Bookworm with the Raspberri Pi Desktop") and it still retained WiFi connectivity after all upgrades.

→ Raspberry Pi OS: it seems to work both with 64-bit and 32-bit

beamsi, 2024-07-25:
I have the same problem too on RaspberryPi 4. Just updated an hour ago and cannot get wifi working again. But ethernet is working fine. I don't know the version before but I updated it last time in the last two months

This is the output of wpa_supplicant. My UniFi-AP
```
Jul 25 17:50:03 pikvm wpa_supplicant[2193]: nl80211: send_event_marker failed: Source based routing not supported
Jul 25 17:50:03 pikvm wpa_supplicant[2193]: wlan0: CTRL-EVENT-DISCONNECTED bssid=xx:xx:xx:xx:xx reason=3 locally_generated=1
Jul 25 17:50:03 pikvm wpa_supplicant[2193]: wlan0: BSSID xx:xx:xx:xx:xx ignore list count incremented to 2, ignoring for 10 seconds
Jul 25 17:50:03 pikvm wpa_supplicant[2193]: wlan0: CTRL-EVENT-SSID-TEMP-DISABLED id=0 ssid="my personal wifi" auth_failures=7 duration=140 reason=CONN_FAILED
Jul 25 17:50:03 pikvm wpa_supplicant[2193]: wlan0: CTRL-EVENT-DSCP-POLICY clear_all
Jul 25 17:50:03 pikvm wpa_supplicant[2193]: wlan0: CTRL-EVENT-REGDOM-CHANGE init=CORE type=WORLD
Jul 25 17:50:03 pikvm wpa_supplicant[2193]: wlan0: CTRL-EVENT-REGDOM-CHANGE init=CORE type=WORLD
```

mdevaev, 2024-07-25:
We don't keep old version. By the way, we didn't perform kernel update and firmware for 1-2 month. Therefore, I am surprised by your message that wifi is broken.

kusmi, 2024-07-26:
> Another one who has the same problem. When installing a new installation there are no problems, but when I run pikvm-update directly the wifi access is lost. Waiting to see if we can find a solution together.

you also run pikvm on a Raspberry Pi Zero 2 W (DYI kit)? or another Raspberry Pi configuration?

zuozhehao, 2024-07-26:
I also encountered the same problem, Raspberry Pi 4b.
I just installed it normally, but after executing `pikvm-update`, the wireless one cannot be used.

kiwinh0, 2024-07-26:
> you also run pikvm on a Raspberry Pi Zero 2 W (DYI kit)? or another Raspberry Pi configuration?

On Raspberry pi 4B

mdevaev, 2024-07-26:
wpa_supplicant is broken. Fixed version will be available after few hours: https://archlinuxarm.org/packages/armv7h/wpa_supplicant

mdevaev, 2024-07-27:
Fixed version still not working. I [reported](https://archlinuxarm.org/forum/viewtopic.php?f=15&t=17007&p=72789) about the problem. Hotfix now available in `pikvm-update`.

closed 2024-07-27 by mdevaev

kusmi, 2024-07-27:
Can confirm - setting up new an do a full `pikvm-update` keeps my Wifi running. So I can now access my server again :-) thanks!