pikvm/pikvm #1377: Keyboard Lock aren't properly working on MacOS Host using either Chrome or Edge.

rudironsonijr, 2024-07-31:
**Describe the bug**
Keyboard Lock aren't properly working on MacOS Host using either Chrome or Edge.

**To Reproduce**
Steps to reproduce the behavior, like:
1. Setup Keyboard Lock on Chrome or Edge flags
2. Install PiKVM as an App (Chrome or Edge)
3. Open the app
4. Put your KVM window in full screen
5. Test the Key Combos

**Expected behavior**
Key combinations like Cmd+Tab, Cmd+Space and other MacOS specific combos aren't being grabber by PiKVM Web Interface.

**Screenshots**
[image]

[image]

[image]

[image]

[image]

**Desktop (please complete the following information):**
 - OS: MacOS Sonoma and MacOS Sequoia
 - Browser: Chrome and Edge
 - Version 127.0.2651.74
 - ~VNC client (if used)~

**PiKVM info:**
  - Raspberry Pi 4 Model B Rev 1.5
  - PiKVM platform v3 (official)
  - kvmd 4.3-1
  - kvmd-fan 0.31-1
  - kvmd-oled 0.27-1
  - kvmd-platform-v3-hdmi-rpi4 4.3-1
  - kvmd-webterm 0.50-1
  - ustreamer 6.12-1
  - Linux kernel: Linux pikvm 6.6.21-4-rpi #1 SMP Wed Mar 13 20:36:42 UTC 2024 armv7l GNU/Linux

**Additional context**
~Add any other context about the problem here.~

rudironsonijr, 2024-08-05:
@mdevaev, may you please help with this?

mdevaev, 2024-11-07:
Found a similar problem: https://github.com/novnc/noVNC/issues/1799

Yeah, it can't be fixed on our side properly, we will have to wait for the fix in the browsers.

closed 2024-11-07 by mdevaev