pikvm/pikvm #1382: Tests fail if `/etc/kvmd/main.yaml` does not exist

rhoriguchi, 2024-08-12:
**Describe the bug**
I'm trying to package `kvmd` for nix but I'm facing the issue that tests cannot run if no `/etc/kvmd/main.yaml` exists even if the tests have nothing to do with that file.

<details>
    <summary>Example error</summary>

    ___________________________________ test_ok ____________________________________

    arg = '/etc/kvmd/main.yaml', type = 'reg'
    name = 'absolute path to existent file'

        def valid_abs_path(arg: Any, type: str="", name: str="") -> str:  # pylint: disable=redefined-builtin
            if type:
                if not name:
                    name = f"absolute path to existent {type}"
                type = {
                    "file": "reg",
                    "dir": "dir",
                    "link": "lnk",
                    "sock": "sock",
                    "fifo": "fifo",
                    "char": "chr",
                    "block": "blk",
                }[type]
            else:
                if not name:
                    name = "absolute path"
        
            arg = os.path.abspath(valid_stripped_string_not_empty(arg, name))
        
            if type:
                try:
    >               st = os.stat(arg)
    E               FileNotFoundError: [Errno 2] No such file or directory: '/etc/kvmd/main.yaml'

    kvmd/validators/os.py:57: FileNotFoundError

    During handling of the above exception, another exception occurred:

    self = ArgumentParser(prog='kvmd-cleanup', usage=None, description='Kill KVMD and clear resources', formatter_class=<class 'argparse.ArgumentDefaultsHelpFormatter'>, conflict_handler='error', add_help=True)
    action = _StoreAction(option_strings=['-c', '--config'], dest='config', nargs=None, const=None, default='/etc/kvmd/main.yaml', ...unction valid_abs_file at 0x7ffff54b2520>, choices=None, required=False, help='Set config file path', metavar='<file>')
    arg_string = '/etc/kvmd/main.yaml'

        def _get_value(self, action, arg_string):
            type_func = self._registry_get('type', action.type, action.type)
            if not callable(type_func):
                msg = _('%r is not callable')
                raise ArgumentError(action, msg % type_func)
        
            # convert the value to the appropriate type
            try:
    >           result = type_func(arg_string)

    /nix/store/l014xp1qxdl6gim3zc0jv3mpxhbp346s-python3-3.12.4/lib/python3.12/argparse.py:2565: 
    _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
    kvmd/validators/os.py:68: in valid_abs_file
        return valid_abs_path(arg, type="file", name=name)
    kvmd/validators/os.py:59: in valid_abs_path
        raise_error(arg, f"{name}: {err}")
    _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 

    arg = '/etc/kvmd/main.yaml'
    name = "absolute path to existent file: [Errno 2] No such file or directory: '/etc/kvmd/main.yaml'"
    hide = False

        def raise_error(arg: Any, name: str, hide: bool=False) -> NoReturn:
            arg_str = " "
            if not hide:
                arg_str = (f" {arg!r} " if isinstance(arg, (str, bytes)) else f" '{arg}' ")
    >       raise ValidatorError(f"The argument{arg_str}is not a valid {name}")
[... 182 lines cut ...]
    E       kvmd.validators.ValidatorError: The argument '/root' is not a valid absolute path to existent dir: [Errno 2] No such file or directory: '/root'

    kvmd/validators/__init__.py:47: ValidatorError
</details>

**To Reproduce**
Steps to reproduce the behavior, like:
1. checkout https://github.com/rhoriguchi/nixpkgs/commit/8f099ebac3cb61287cd4239638d3b580b5b962d5
2. run `nix-build -A python312Packages.kvmd`
3. find line with `For full logs, run 'nix log /nix/store/0r4a48527pp10x85jrdzga6grx2pbnq1-python3.12-kvmd-4.2.drv'.`
4. [log output](https://github.com/user-attachments/files/16576676/logs.txt)

**Expected behavior**
Tests should run without `/etc/kvmd/main.yaml`.

**Desktop (please complete the following information):**
 - OS: NixOS

**PiKVM info:**
  - KVMD version: 4.2
  - uStreamer version: 6.12
  - Linux kernel: 6.10.2

**Additional context**
Tests should not depend on files in the environment.

mdevaev, 2024-08-12:
Hello. Tests can only work inside the container because many involve changes to the OS. They are not supposed to be launched when creating a package. So the main.yaml is not the only thing you will encounter if you try to do this. I recommend skipping the tests.

rhoriguchi, 2024-08-12:
Hi @mdevaev, thx for the reply. I try to get as many tests to run as possible to give me more confidence that an update won't break anything. Pytest does offer a fixture to test stuff that requires paths <https://docs.pytest.org/en/stable/how-to/tmp_path.html>. If you don't wish to address this, please close the issue and I'll just exclude all tests that fail.

I've already excluded this test since it calls `chpasswd` and `chpasswd`.
```
  disabledTestPaths = [
    # uses fixture that calls `chpasswd` and `useradd`
    "testenv/tests/plugins/auth/test_pam.py"
  ];
  ```

mdevaev, 2024-08-12:
I checked it again. This file is required for some integration tests, as well as the rest of the configuration in testenv, so the fixture will not help here.

closed 2024-08-12 by mdevaev