pikvm/pikvm #1384: piKVM does not remotely display the computer screen which it is connected to.

munchify, 2024-08-18:
I have 7 servers (handbuilt) which are connected to piKVMs for remote operation. These worked for the better part of a year until I had to replace a couple of the GPUs for maintenance. However, after powering the servers back on, 5 of the 7 KVMs display "no signal" - they are still functional and can be connected to remotely, however they fail to show the computer screen and allow remote operation.

These are the steps I've taken to fix it:
- Power on and off
- Put the old GPUs back in the exact same configuration as before
- Update the software on the SD cards of the KVMs
- Disconnect and reconnect KVMs to the server
- Connected the server directly to a monitor screen to make sure the server is still giving a correct output, which it is
- Use a USB-C cable to give extra power to the KVMs to make sure that power was not an issue - at some point one of the error messages was a insufficient power however I was using a 60W (also used a POE booster with ethernet cable) so that is unlikely to be the case

The part that I'm most confused about is that the two functional servers are wired exactly the same as the others, however they are the only two that work. Would greatly appreciate any help.

Desktop:
 - OS: Ubunute 24 LTS (Server edition)
 
PiKVM info:
  - RPi 4-B, KVM-A8
  - Connected via HDMI to srever
  - Linux kernel: `uname -a`

mdevaev, 2024-08-18:
> Update the software on the SD cards of the KVMs

I suppose this step is not necessary.

Which OS do you have on your servers?

shroomsnorter, 2024-08-18:
Running Ubuntu 22.04.4 Live Server (amd64) on five older ones and Ubuntu 24.04 Live Server (amd64) on two newer ones.

Two of the older ones are still working

mdevaev, 2024-08-18:
Have you tried the BIOS access? Does it show video?

shroomsnorter, 2024-08-18:
Thanks for the advice, I'll try it out tomorrow as soon as I get access to the servers and update.

mdevaev, 2024-08-29:
Sup?

mdevaev, 2024-08-30:
Closing due no feedback.

closed 2024-08-30 by mdevaev