pikvm/pikvm #1386: No video / Janus in crash loop

briha, 2024-08-22:
**Describe the bug**
No KVM video as Janus is not reachable. When checking Janus' log, it is in a crash loop.

**To Reproduce**
Steps to reproduce the behavior, like:
1. Start Raspberry Pi running PiKVM
2. Attempt to use web interface KVM
3. No video visible.

**Expected behavior**
Video visible.

**Desktop (please complete the following information):**
 - OS: Windows 11
 - Browser: Firefox, Chrome
 - Version: 129.0.1, 124.0.6314.0
 - VNC client (if used): N/A

**PiKVM info:**
  - Raspberry Pi board version: Zero 2 W
  - PiKVM platform [e.g. v2-hdmi]: v2-hdmi
  - Video capture type [e.g. CSI bridge]: CSI bridge
  - KVMD version: `pacman -Q | grep kvmd`: 4.6-1
  - uStreamer version: `pacman -Q | grep ustreamer`: 6.13-1
  - Linux kernel: `uname -a`:  6.6.45-2-rpi

**Additional context**
Browser log:
```Firefox can’t establish a connection to the server at wss://192.168.1.23/janus/ws. [janus.js:98:50](https://192.168.1.23/share/js/kvm/janus.js)
Error connecting to the Janus WebSockets server... wss://192.168.1.23/janus/ws [janus.js:875:12](https://192.168.1.23/share/js/kvm/janus.js)
[08:29:32.987] LOG/ERROR -- Stream [Janus]: Error connecting to the Janus WebSockets server: Is the server down? [tools.js:380:11](https://192.168.1.23/share/js/tools.js)
[08:29:38.746] LOG/INFO -- Stream [Janus]: Starting Janus ... [tools.js:380:11](https://192.168.1.23/share/js/tools.js)
Library initialized: true [janus.js:456:8](https://192.168.1.23/share/js/kvm/janus.js)
Using WebSockets to contact Janus: wss://192.168.1.23/janus/ws [janus.js:476:10](https://192.168.1.23/share/js/kvm/janus.js)
Firefox can’t establish a connection to the server at wss://192.168.1.23/janus/ws. [janus.js:98:50](https://192.168.1.23/share/js/kvm/janus.js)
Error connecting to the Janus WebSockets server... wss://192.168.1.23/janus/ws [janus.js:875:12](https://192.168.1.23/share/js/kvm/janus.js)
[08:30:33.103] LOG/ERROR -- Stream [Janus]: Error connecting to the Janus WebSockets server: Is the server down? [tools.js:380:11](https://192.168.1.23/share/js/tools.js)
[08:30:38.124] LOG/INFO -- Stream [Janus]: Starting Janus ... [tools.js:380:11](https://192.168.1.23/share/js/tools.js)
Library initialized: true [janus.js:456:8](https://192.168.1.23/share/js/kvm/janus.js)
Using WebSockets to contact Janus: wss://192.168.1.23/janus/ws [janus.js:476:10](https://192.168.1.23/share/js/kvm/janus.js)
Firefox can’t establish a connection to the server at wss://192.168.1.23/janus/ws. [janus.js:98:50](https://192.168.1.23/share/js/kvm/janus.js)
Error connecting to the Janus WebSockets server... wss://192.168.1.23/janus/ws [janus.js:875:12](https://192.168.1.23/share/js/kvm/janus.js)
[08:31:33.213] LOG/ERROR -- Stream [Janus]: Error connecting to the Janus WebSockets server: Is the server down? [tools.js:380:11](https://192.168.1.23/share/js/tools.js)
[08:31:38.287] LOG/INFO -- Stream [Janus]: Starting Janus ... [tools.js:380:11](https://192.168.1.23/share/js/tools.js)
Library initialized: true [janus.js:456:8](https://192.168.1.23/share/js/kvm/janus.js)
Using WebSockets to contact Janus: wss://192.168.1.23/janus/ws [janus.js:476:10](https://192.168.1.23/share/js/kvm/janus.js)
GET
wss://192.168.1.23/janus/ws
NS_ERROR_WEBSOCKET_CONNECTION_REFUSED

Firefox can’t establish a connection to the server at wss://192.168.1.23/janus/ws. [janus.js:98:50](https://192.168.1.23/share/js/kvm/janus.js)
Error connecting to the Janus WebSockets server... wss://192.168.1.23/janus/ws [janus.js:875:12](https://192.168.1.23/share/js/kvm/janus.js)
[08:32:34.010] LOG/ERROR -- Stream [Janus]: Error connecting to the Janus WebSockets server: Is the server down? [tools.js:380:11](https://192.168.1.23/share/js/tools.js)
[08:32:39.065] LOG/INFO -- Stream [Janus]: Starting Janus ... [tools.js:380:11](https://192.168.1.23/share/js/tools.js)
Library initialized: true [janus.js:456:8](https://192.168.1.23/share/js/kvm/janus.js)
Using WebSockets to contact Janus: wss://192.168.1.23/janus/ws [janus.js:476:10](https://192.168.1.23/share/js/kvm/janus.js)
GET
wss://192.168.1.23/janus/ws
NS_ERROR_WEBSOCKET_CONNECTION_REFUSED

Firefox can’t establish a connection to the server at wss://192.168.1.23/janus/ws. [janus.js:98:50](https://192.168.1.23/share/js/kvm/janus.js)
Error connecting to the Janus WebSockets server... wss://192.168.1.23/janus/ws [janus.js:875:12](https://192.168.1.23/share/js/kvm/janus.js)
[08:33:34.259] LOG/ERROR -- Stream [Janus]: Error connecting to the Janus WebSockets server: Is the server down? [tools.js:380:11](https://192.168.1.23/share/js/tools.js)
```

Janus log:
```
Aug 22 08:30:05 pikvm kvmd-janus[1295]:   File "/usr/lib/python3.12/site-packages/kvmd/apps/janus/runner.py", line 151, in __janus_task_loop
Aug 22 08:30:05 pikvm kvmd-janus[1295]:     raise RuntimeError("Janus unexpectedly died")
Aug 22 08:30:05 pikvm kvmd-janus[1295]: RuntimeError: Janus unexpectedly died
[... 25 lines cut ...]
Aug 22 08:30:11 pikvm kvmd-janus[1295]:   File "/usr/lib/python3.12/site-packages/kvmd/apps/janus/runner.py", line 151, in __janus_task_loop
Aug 22 08:30:11 pikvm kvmd-janus[1295]:     raise RuntimeError("Janus unexpectedly died")
Aug 22 08:30:11 pikvm kvmd-janus[1295]: RuntimeError: Janus unexpectedly died
Aug 22 08:30:12 pikvm kvmd-janus[1295]: kvmd.apps.janus.runner            INFO --- Started Janus pid=2148: /usr/bin/janus --disable-colors --plugins-folder=/usr/lib/ustreamer/janus --configs-folder=/etc/kvmd/janus --interface=192.168.1.23 --stun-server=>
Aug 22 08:30:13 pikvm kvmd-janus[1295]: kvmd.apps.janus.runner            INFO --- => Janus commit: not-a-git-repo
Aug 22 08:30:13 pikvm kvmd-janus[1295]: kvmd.apps.janus.runner            INFO --- => Compiled on:  Sun Apr  7 20:02:58 UTC 2024
Aug 22 08:30:13 pikvm kvmd-janus[1295]: kvmd.apps.janus.runner            INFO --- => Logger plugins folder: /usr/lib/janus/loggers
Aug 22 08:30:13 pikvm kvmd-janus[1295]: kvmd.apps.janus.runner            INFO --- => ---------------------------------------------------
Aug 22 08:30:13 pikvm kvmd-janus[1295]: kvmd.apps.janus.runner            INFO --- => Starting Meetecho Janus (WebRTC Server) v0.14.3
Aug 22 08:30:13 pikvm kvmd-janus[1295]: kvmd.apps.janus.runner            INFO --- => ---------------------------------------------------
Aug 22 08:30:13 pikvm kvmd-janus[1295]: kvmd.apps.janus.runner            INFO --- => Checking command line arguments...
Aug 22 08:30:13 pikvm kvmd-janus[1295]: kvmd.apps.janus.runner            INFO --- => Debug/log level is 4
Aug 22 08:30:13 pikvm kvmd-janus[1295]: kvmd.apps.janus.runner            INFO --- => Debug/log timestamps are disabled
Aug 22 08:30:13 pikvm kvmd-janus[1295]: kvmd.apps.janus.runner            INFO --- => Debug/log colors are disabled
Aug 22 08:30:13 pikvm kvmd-janus[1295]: kvmd.apps.janus.runner            INFO --- => Using 192.168.1.23 as local IP...
Aug 22 08:30:13 pikvm kvmd-janus[1295]: kvmd.apps.janus.runner            INFO --- => Token based authentication disabled
Aug 22 08:30:13 pikvm kvmd-janus[1295]: kvmd.apps.janus.runner            INFO --- => Initializing recorder code
Aug 22 08:30:13 pikvm kvmd-janus[1295]: kvmd.apps.janus.runner            INFO --- => RTP port range: 20000 -- 40000
Aug 22 08:30:13 pikvm kvmd-janus[1295]: kvmd.apps.janus.runner            INFO --- => Initializing ICE stuff (Full mode, ICE-TCP candidates disabled, half-trickle, IPv6 support enabled)
Aug 22 08:30:13 pikvm kvmd-janus[1295]: kvmd.apps.janus.runner            INFO --- => ICE port range: 20000-40000
Aug 22 08:30:13 pikvm kvmd-janus[1295]: kvmd.apps.janus.runner            INFO --- => [WARN] mDNS resolution disabled, .local candidates will be ignored
Aug 22 08:30:13 pikvm kvmd-janus[1295]: kvmd.apps.janus.runner            INFO --- => STUN server to use: stun.l.google.com:19302
Aug 22 08:30:13 pikvm kvmd-janus[1295]: kvmd.apps.janus.runner            INFO --- => >> 2001:4860:4864:5:8000::1:19302 (IPv6)
Aug 22 08:30:13 pikvm kvmd-janus[1295]: kvmd.apps.janus.runner            INFO --- => Testing STUN server: message is of 20 bytes
```

mdevaev, 2024-08-22:
Hello. Please try to replace sd card, and check again.

briha, 2024-08-23:
That didn't do anything. Adding `ignore_unreachable_ice_server = true` to `/etc/kvmd/janus/janus.jcfg` in the `nat` section did the trick though. Any idea why that is needed in the first place?

mdevaev, 2024-08-23:
I see. Why `stun.l.google.com:19302` doesn't available from your network? It is required for using webrtc video from external network: https://docs.pikvm.org/webrtc/#how-its-working

briha, 2024-08-23:
I am not sure. As I don't have any intentions of using it from an external network, disabling the check shouldn't pose a problem, correct?

mdevaev, 2024-08-23:
For this case you can use kvmd-janus-static, check the doc that I gave you.
I don't recommend to change default configs.

briha, 2024-08-26:
The config for `kvmd-janus-static` seems to be the exact same that I already modified -- what's the difference between `kvmd-janus` and `kvmd-janus-static`?

mdevaev, 2024-08-26:
It described in the doc that I gave you.

[image]

briha, 2024-08-26:
I understand that part. What I fail to understand, is the difference between the regular `kvmd-janus` and the `-static` version. I changed the exact config described in that paragraph, yet that paragraph also tells me to activate the `-static` Janus service, while the config change itself also applies to the non-static `kvmd-janus`. Hence the question: What is the difference between `kvmd-janus-static` and `kvmd-janus`, as clearly both seem to use the same config file?

mdevaev, 2024-08-26:
`kvmd-janus-static` just runs janus. `kvmd-janus` runs a runner for janus that follows network config. In case of static you should follow it manually.

mdevaev, 2024-09-13:
@briha do you have more `kvmd-janus` logs? I want to understand what preceded the crash because in fact if STUN is not available it should have started anyway.

briha, 2024-09-16:
I uncommented the ignore_unreachable_ice_server and restarted the janus service, this is the log (after that the crash loop just repeates indefinitively):
```Sep 16 17:39:38 pikvm systemd[1]: Stopping PiKVM - Janus WebRTC Gateway...
Sep 16 17:39:38 pikvm kvmd-janus[489]: kvmd.apps.janus.runner            INFO --- Stopping Janus ...
Sep 16 17:39:43 pikvm kvmd-janus[489]: kvmd.apps.janus.runner            INFO --- Process killed: retcode=0
Sep 16 17:39:43 pikvm kvmd-janus[489]: kvmd.apps.janus.runner            INFO --- Bye-bye
Sep 16 17:39:44 pikvm systemd[1]: kvmd-janus.service: Deactivated successfully.
Sep 16 17:39:44 pikvm systemd[1]: Stopped PiKVM - Janus WebRTC Gateway.
Sep 16 17:39:44 pikvm systemd[1]: kvmd-janus.service: Consumed 14min 41.433s CPU time.
Sep 16 17:39:44 pikvm systemd[1]: Started PiKVM - Janus WebRTC Gateway.
Sep 16 17:39:44 pikvm kvmd-janus[25897]: kvmd.apps.janus.runner            INFO --- Starting Janus Runner ...
Sep 16 17:39:44 pikvm kvmd-janus[25897]: kvmd.apps.janus.runner            INFO --- Probbing the network first time ...
Sep 16 17:39:44 pikvm kvmd-janus[25897]: kvmd.apps.janus.runner            INFO --- Got new _Netcfg(nat_type='Full Cone NAT', src_ip='xx.xx.xx.xx', ext_ip='xx.xx.xx.xx', stun_host='stun.l.google.com', stun_port=19302)
Sep 16 17:39:44 pikvm kvmd-janus[25897]: kvmd.apps.janus.runner            INFO --- Starting Janus ...
Sep 16 17:39:44 pikvm kvmd-janus[25897]: kvmd.apps.janus.runner            INFO --- Started Janus pid=25908: /usr/bin/janus --disable-colors --plugins-folder=/usr/lib/ustreamer/janus --configs-folder=/etc/kvmd/janus --interface=xx.xx.xx.xx --s>
Sep 16 17:39:44 pikvm kvmd-janus[25897]: kvmd.apps.janus.runner            INFO --- => Janus commit: not-a-git-repo
Sep 16 17:39:44 pikvm kvmd-janus[25897]: kvmd.apps.janus.runner            INFO --- => Compiled on:  Sun Apr  7 20:02:58 UTC 2024
Sep 16 17:39:44 pikvm kvmd-janus[25897]: kvmd.apps.janus.runner            INFO --- => Logger plugins folder: /usr/lib/janus/loggers
Sep 16 17:39:44 pikvm kvmd-janus[25897]: kvmd.apps.janus.runner            INFO --- => ---------------------------------------------------
Sep 16 17:39:44 pikvm kvmd-janus[25897]: kvmd.apps.janus.runner            INFO --- => Starting Meetecho Janus (WebRTC Server) v0.14.3
Sep 16 17:39:44 pikvm kvmd-janus[25897]: kvmd.apps.janus.runner            INFO --- => ---------------------------------------------------
Sep 16 17:39:44 pikvm kvmd-janus[25897]: kvmd.apps.janus.runner            INFO --- => Checking command line arguments...
Sep 16 17:39:44 pikvm kvmd-janus[25897]: kvmd.apps.janus.runner            INFO --- => Debug/log level is 4
Sep 16 17:39:44 pikvm kvmd-janus[25897]: kvmd.apps.janus.runner            INFO --- => Debug/log timestamps are disabled
Sep 16 17:39:44 pikvm kvmd-janus[25897]: kvmd.apps.janus.runner            INFO --- => Debug/log colors are disabled
Sep 16 17:39:44 pikvm kvmd-janus[25897]: kvmd.apps.janus.runner            INFO --- => Using xx.xx.xx.xx as local IP...
Sep 16 17:39:44 pikvm kvmd-janus[25897]: kvmd.apps.janus.runner            INFO --- => Token based authentication disabled
Sep 16 17:39:44 pikvm kvmd-janus[25897]: kvmd.apps.janus.runner            INFO --- => Initializing recorder code
Sep 16 17:39:44 pikvm kvmd-janus[25897]: kvmd.apps.janus.runner            INFO --- => RTP port range: 20000 -- 40000
Sep 16 17:39:44 pikvm kvmd-janus[25897]: kvmd.apps.janus.runner            INFO --- => Initializing ICE stuff (Full mode, ICE-TCP candidates disabled, half-trickle, IPv6 support enabled)
Sep 16 17:39:44 pikvm kvmd-janus[25897]: kvmd.apps.janus.runner            INFO --- => ICE port range: 20000-40000
Sep 16 17:39:44 pikvm kvmd-janus[25897]: kvmd.apps.janus.runner            INFO --- => [WARN] mDNS resolution disabled, .local candidates will be ignored
Sep 16 17:39:44 pikvm kvmd-janus[25897]: kvmd.apps.janus.runner            INFO --- => STUN server to use: stun.l.google.com:19302
Sep 16 17:39:44 pikvm kvmd-janus[25897]: kvmd.apps.janus.runner            INFO --- => >> 2001:4860:4864:5:8000::1:19302 (IPv6)
Sep 16 17:39:44 pikvm kvmd-janus[25897]: kvmd.apps.janus.runner            INFO --- => Testing STUN server: message is of 20 bytes
Sep 16 17:39:50 pikvm kvmd-janus[25897]: kvmd.apps.janus.runner            INFO --- => [FATAL] [ice.c:janus_ice_test_stun_server:1137] No response to our STUN BINDING test
Sep 16 17:39:50 pikvm kvmd-janus[25897]: kvmd.apps.janus.runner            INFO --- => [FATAL] [janus.c:main:5061] Invalid STUN address stun.l.google.com:19302
Sep 16 17:39:50 pikvm kvmd-janus[25897]: kvmd.apps.janus.runner           ERROR --- Unexpected Janus error: pid=25908
Sep 16 17:39:50 pikvm kvmd-janus[25897]: Traceback (most recent call last):
Sep 16 17:39:50 pikvm kvmd-janus[25897]:   File "/usr/lib/python3.12/site-packages/kvmd/apps/janus/runner.py", line 151, in __janus_task_loop
Sep 16 17:39:50 pikvm kvmd-janus[25897]:     raise RuntimeError("Janus unexpectedly died")
Sep 16 17:39:50 pikvm kvmd-janus[25897]: RuntimeError: Janus unexpectedly died
Sep 16 17:39:51 pikvm kvmd-janus[25897]: kvmd.apps.janus.runner            INFO --- Started Janus pid=25917: /usr/bin/janus --disable-colors --plugins-folder=/usr/lib/ustreamer/janus --configs-folder=/etc/kvmd/janus --interface=xx.xx.xx.xx --s>
Sep 16 17:39:51 pikvm kvmd-janus[25897]: kvmd.apps.janus.runner            INFO --- => Janus commit: not-a-git-repo
[... cut ...]
Sep 16 17:39:56 pikvm kvmd-janus[25897]: kvmd.apps.janus.runner           ERROR --- Unexpected Janus error: pid=25917
Sep 16 17:39:56 pikvm kvmd-janus[25897]: Traceback (most recent call last):
Sep 16 17:39:56 pikvm kvmd-janus[25897]:   File "/usr/lib/python3.12/site-packages/kvmd/apps/janus/runner.py", line 151, in __janus_task_loop
Sep 16 17:39:56 pikvm kvmd-janus[25897]:     raise RuntimeError("Janus unexpectedly died")
Sep 16 17:39:56 pikvm kvmd-janus[25897]: RuntimeError: Janus unexpectedly died
```

mdevaev, 2024-09-16:
Got it, thank you. Do you have a working IPv6 in your network?

briha, 2024-09-16:
I just checked: apparently not. Can `stun.l.google.com:19302` not be accessed through IPv4?

mdevaev, 2024-09-16:
It can. I'm interested why Janus resolved it as IPv6

mdevaev, 2024-09-16:
Please run on PiKVM:
```
python -c 'import socket, pprint; pprint.pprint(socket.getaddrinfo("stun.l.google.com", 0))'
```

briha, 2024-09-16:
```
# python -c 'import socket, pprint; pprint.pprint(socket.getaddrinfo("stun.l.google.com", 0))'
[(<AddressFamily.AF_INET6: 10>,
  <SocketKind.SOCK_STREAM: 1>,
  6,
  '',
  ('2001:4860:4864:5:8000::1', 0, 0, 0)),
 (<AddressFamily.AF_INET6: 10>,
  <SocketKind.SOCK_DGRAM: 2>,
  17,
  '',
  ('2001:4860:4864:5:8000::1', 0, 0, 0)),
 (<AddressFamily.AF_INET6: 10>,
  <SocketKind.SOCK_RAW: 3>,
  0,
  '',
  ('2001:4860:4864:5:8000::1', 0, 0, 0)),
 (<AddressFamily.AF_INET: 2>,
  <SocketKind.SOCK_STREAM: 1>,
  6,
  '',
  ('74.125.250.129', 0)),
 (<AddressFamily.AF_INET: 2>,
  <SocketKind.SOCK_DGRAM: 2>,
  17,
  '',
  ('74.125.250.129', 0)),
 (<AddressFamily.AF_INET: 2>,
  <SocketKind.SOCK_RAW: 3>,
  0,
  '',
```

mdevaev, 2024-09-16:
In the log:
```
Got new _Netcfg(nat_type='Full Cone NAT', src_ip='xx.xx.xx.xx', ext_ip='xx.xx.xx.xx', stun_host='stun.l.google.com', stun_port=19302)
```
src_ip and ext_ip are IPv4, right?

briha, 2024-09-16:
yes, they are

mdevaev, 2024-09-16:
Okay. I'll try to make some workaround.

mdevaev, 2024-09-17:
Please perform `pikvm-update` and try `kvmd-janus` without `ignore_unreachable_ice_server`.

briha, 2024-09-17:
Yes, that did it. However, updating was a huge pain.
Apparently, there are some broken mirrors in the automatically select Arch Linux Arm mirrors (or this is another case of an IPv4 issue). Only when I added
```
 Server = http://de3.mirror.archlinuxarm.org/$arch/$repo
```
instead of the default 
```
Server = http://mirror.archlinuxarm.org/$arch/$repo
```
would `pikvm-update` go through.

Before that, it would always manage to download a few packages and then fail on one of them (presumably, when a bad mirror was selected). An example of what that looked like:
```
# pikvm-update
+ trap on_error ERR
+ _yes='--noconfirm --ask=4 --overwrite \*'
+ rw
+ mount -o remount,rw /
+ mount -o remount,rw /boot
+ set +x
=== PiKVM is in Read-Write mode ===
+ pacman -Syy
:: Synchronizing package databases...
 core                                         238.8 KiB   146 KiB/s 00:02 [##########################################] 100%
 extra                                          9.0 MiB   594 KiB/s 00:16 [##########################################] 100%
 community                                     45.0   B   236   B/s 00:00 [##########################################] 100%
 alarm                                         94.9 KiB   316 KiB/s 00:00 [##########################################] 100%
 aur                                            9.3 KiB  33.3 KiB/s 00:00 [##########################################] 100%
 pikvm                                         10.9 KiB  31.2 KiB/s 00:00 [##########################################] 100%
++ pacman -S -u --print-format %n
++ grep -v pikvm-os-updater
++ wc -l
+ '[' 32 -eq 0 ']'
+ rm -f /var/cache/pacman/pkg/perl-5.40.0-1-armv7h.pkg.tar.xz /var/cache/pacman/pkg/perl-5.40.0-1-armv7h.pkg.tar.xz.sig /var/cache/pacman/pkg/systemd-256.6-1-armv7h.pkg.tar.xz.part
+ '[' -z '' ']'
++ pacman -S --needed --print-format %n pikvm-os-updater
++ wc -l
+ '[' 0 -ne 0 ']'
+ grep -q '^C.UTF-8 UTF-8' /etc/locale.gen
+ for _pkg in rpi-eeprom rpi4-eeprom
+ pacman -Q rpi-eeprom
+ for _pkg in rpi-eeprom rpi4-eeprom
+ pacman -Q rpi4-eeprom
+ pacman -Qi python-ajsonrpc
+ grep Depends
+ grep -q -e 'python<3\.11' -e 'python<3\.12'
error: package 'python-ajsonrpc' was not found
+ pacman -Q raspberrypi-utils libgpiod
+ pacman --noconfirm --ask=4 --overwrite '\*' -Su
:: Starting full system upgrade...
resolving dependencies...
looking for conflicting packages...
:: kvmd-4.16-1 and kvmd-oled-0.27-1 are in conflict. Remove kvmd-oled? [y/N] 

Packages (33) audit-4.0.2-2  base-devel-1-2  bison-3.8.2-8  cryptsetup-2.7.5-1  curl-8.10.0-1  dtc-1.7.1-1  dysk-2.9.1-1
              expat-2.6.3-2  gc-8.2.8-2  gettext-0.22.5-2  hwdata-0.387-1  iputils-20240905-1  kvmd-4.16-1
              kvmd-oled-0.27-1 [removal]  kvmd-platform-v2-hdmi-rpi4-4.16-1  ldns-1.8.4-1  libisl-0.27-1  libldap-2.6.8-2
              nano-8.2-1  openssl-3.3.2-1  perl-5.40.0-1  perl-error-0.17029-7  perl-mailtools-2.21-9
              perl-timedate-2.33-7  popt-1.19-2  python-urllib3-1.26.20-1  systemd-256.6-1  systemd-libs-256.6-1
              systemd-sysvcompat-256.6-1  texinfo-7.1.1-1  tzdata-2024b-2  ustreamer-6.16-1  wireless-regdb-2024.07.04-2

Total Download Size:    40.52 MiB
Total Installed Size:  160.07 MiB
Net Upgrade Size:        1.86 MiB

:: Proceed with installation? [Y/n] 
:: Retrieving packages...
 perl-5.40.0-1-armv7h                          14.5 MiB  1372 KiB/s 00:11 [##########################################] 100%
 systemd-256.6-1-armv7h                         7.5 MiB  2038 KiB/s 00:04 [##########################################] 100%
 openssl-3.3.2-1-armv7h.pkg.tar.xz failed to download
 Total ( 3/32)                                 40.5 MiB  2.26 MiB/s 00:18 [##########################################] 100%
error: failed retrieving file 'openssl-3.3.2-1-armv7h.pkg.tar.xz' from mirror.archlinuxarm.org : Recv failure: Connection reset by peer
warning: failed to retrieve some files
error: failed to commit transaction (download library error)
Errors occurred, no packages were upgraded.
++ on_error
++ set +x
==============================================================
      An unexpected error occurred during the update.
==============================================================

==============================================================
Please note that the filesystem now remain in Read-Write mode.
       A reboot is necessary to make it Read-Only again.
The reboot can be performed later using the 'reboot' command.
==============================================================
++ set -x

```
It might be nice to tell `pacman`/`wget` to re-attempt a download when failed (or some other fix here). Changing the `Xfercommand` to do that myself wasn't possible, as this version of [pacman just segfaults](https://bbs.archlinux.org/viewtopic.php?pid=2157961#p2157961) when that is attempted.

mdevaev, 2024-09-17:
Reflash and update then.

mdevaev, 2024-09-18:
UPD: You can also try some other mirror, replace hostname in /etc/pacman.d/mirrorlist to de3.mirror.archlinuxarm.org, for example.

briha, 2024-09-18:
```
Server = http://de3.mirror.archlinuxarm.org/$arch/$repo
```
Worked, as I mentioned before.

The changes you made worked too, it was successfully able to connect to the STUN service using IPv4.

I just mentioned the update issue as it was very annoying and would be nice to have fixed as well (i.e., try again when connection is reset instead of completely giving up).

mdevaev, 2024-09-18:
I don't think I can fix it on my side. Let's wait for the upstream fix.

As for the Janus - glad to hear that this is working now.

closed 2024-09-18 by mdevaev