pikvm/pikvm #1391: Shortcuts Key Combo "ScrLk + ScrLk" and "left Ctrl + left Ctrl"?

Josho-SAI, 2024-08-26:
Hi mdevaev, I'm using a PiKVM connected to a [TESmart KVM Switcher](https://www.amazon.com.au/TESmart-Inputs-Control-Computers-Servers/dp/B07K1X1BPZ). The HDMI switcher hosts 2 separate USB-A keyboard and mouse inputs, so it doesn't have a working mouse. The servers are all headless servers so no desktop is loaded, plus navigating the server BIOS is easy via keyboard only. No mouse is required

The TESmart HDMI switcher uses a combo of "ScrLk + ScrLk + <keys 1 to 8>" for inputs 1 to 8. 

If, like me, you use a laptop that doesn't host a scroll lock key, the TESmart can change this behaviour to use "left-Ctrl + left-Ctrl + <keys 1 to 8>". 

Can I please submit a request to have both "ScrLk + "ScrLk" and "left-Ctrl + left-Ctrl" shortcut keys added? TIA

Very much looking forward to seeing a shortcut customisable option made available. This is the 2nd of possibly quite a few PiKVM and TESmart HDMI switchers I've set up.

mdevaev, 2024-09-09:
Sup. The menu does not allow to create double keyboard shortcuts :/ But you still have to add another click after the double hotkey. Will this really solve the problem?

Josho-SAI, 2024-09-09:
Hi mdevaev, thanks for your reply. To be honest I wasn't sure this combination of keys would be possible and yes, the last key "1 to 8", would be a manual keyboard strike. Anyway, I hope there is still motivation to develop a custom keyboard hotkey solution. The PiKVM is an excellent OoBM option. I will be continuing to use them for my rack builds.

mdevaev, 2025-10-26:
In the new implementation of #207, I'll add this feature. So I'll close this as a duplicate and add your request to that task.